AlexandreRouma/SDRPlusPlus #620: on linux sdrpp fails to operate in windowed mode in KDE with multiple monitors/screens.

EndlessEden, 2022-01-29:
OS: Linux
Distribution: Arch Linux
Desktop Environment: Plasma (Both X11 and Wayland) | KDE ver:5.23.5, QT ver: 5.15.2
GPU: Nvidia rtx 2070 (Driver: nvidia-495.46)
Package: sdrpp-git (Version: r1071.260651f-1)

When sdrpp starts, it opens fullscreen. Hiding the titlebar and launcherbar and requiring window switching to expose the titlebar. when right-clicking the application icon in the launcherbar task-manager, and going to "more actions" the radial boxes "minimize" and "Fullscreen" are selected.

When deslecting "Fullscreen" and clicking the icon, the application appears on top most screen (non-primary) and is in its proper windowed mode with title bar and border.

However, deselecting or moving the window causes strange events to occur. Including automatic minimising and zooming of the entire window. as if weird scaling is occuring or the "Zoom" desktop effect is being applied. 

This effect persists after program termination and requires the display manager to be restarted or sdrpp to be relaunched.

(i will reply with

AlexandreRouma, 2022-01-29:
I'm not sure I fully understand your issue.
First of all F11 fullscreen is not windowed fullscreen, it takes complete control of the main display (multimonitor not supported yet).
Secondly, the other issues you're having honestly sounds like a broken KDE install, especially since I see you're running arch.  I'm running BSPWM on mine and have no issue with fullscreen.

I'd suggest recording the issue happening, it would be easier to understand what's going on.

AlexandreRouma, 2022-10-10:
Closing as stalled, reopen if more info

closed 2022-10-10 by AlexandreRouma