AlexandreRouma/SDRPlusPlus #622: Sound in Spyserver_module

gvanem, 2022-01-29:
I tried the spyserver-module connecting to `sdr://k2zn.ddns.net:5555`:
* An Airspy HF+ receiver in Rochester, New York.
* Server ver 2.0.1700 (?)
* Sample-rate 912 kS/s.

The sound is totally unintelligible: http://www.watt-32.net/dx/SDRpp-Spyserver-k2zn.ddns.net_port-5555.mp3

I tried different sample rates and sizes, different frequency-bands. But no difference.
Not sure it's an issue with audio on Windows, structure packing in the protocol or how samples are processes. 

Anyone?

AlexandreRouma, 2022-01-29:
Spyserver in SDR++ runs at full IQ because that's the only thing open source clients are allowed to use. You simply don't have enough network bandwidth for it to fit through. Try reducing the bit depth to 8bits (will cause issues that I'm not allowed to fix) an,d drop the samplerate.

gvanem, 2022-01-29:
I have plenty network B/W (100 MBit/s). And the above URL works fine in SDR# (ver.1.0.0.1630) with 32-bit floats at 780 kS/s.

> Spyserver in SDR++ runs at full IQ

And SDR# doesn't? But:
 * at 57 kS/s and `Int16` it sounds perfect!
 * at 114 kS/s and `UInt8` it sounds almost perfect!

*PS*. Is there a way to change the sample-rate w/o disconnecting and reconnecting?

AlexandreRouma, 2022-01-29:
SDR# doesn't run full IQ, it runs FFT&VFO and has some features enabled to make 8bit pcm work.
Your connection isn't the bottleneck, it's the other person's upload speed.

gvanem, 2022-01-29:
> Your connection isn't the bottleneck

God to know. Thanks for clearing up.

closed 2022-01-29 by gvanem