AlexandreRouma/SDRPlusPlus #623: SPYserver module

loewal, 2022-01-29:
Hi all,
Can anyone tell me if SPYserver module is comparable with that function in SDRsharp?
I tried to connect to my own spyserver, the same way as in sdrsharp, but nothing happens.

I read this:
I **tried the spyserver-module connecting to sdr://k2zn.ddns.net:5555:**

• An Airspy HF+ receiver in Rochester, New York.
• Server ver 2.0.1700 (?)
• Sample-rate 912 kS/s.

The sound is totally unintelligible: http://www.watt-32.net/dx/SDRpp-Spyserver-k2zn.ddns.net_port-5555.mp3
How please.....

AlexandreRouma, 2022-01-29:
Uhhh.... https://github.com/AlexandreRouma/SDRPlusPlus/issues/622

AlexandreRouma, 2022-01-29:
I'll also add that connecting isn't enough, you have to press play

loewal, 2022-01-29:
But connecting doesn’t succeed Alexandre.
I filled in that link that Giles used, but there is no connection established.

Could you please give more explanation about how to setup this?

Thank you.

Radiomix2000, 2022-01-29:
As this topic belongs to Spyserver thematic, I have to mention one strange behavior of SDRPP, when connected to spyserver.
The SR setting is 125 000 sps.
Both server and client are in the same LAN.
Connection to server and starting the stream is ok.
But when you are connected for some time, the stream stops accidently. It may happen even after some seconds after starting stream, but can work 15 minutes without stop the stream. The time when it will happen is impossible to determine. Seems to be random event..

AlexandreRouma, 2022-01-29:
@Radiomix2000 please open your own issue with your own specific details

AlexandreRouma, 2022-01-29:
@loewal Can you give more info on your issue, I need to know the version of spyserver, how you're trying to connect and what error message you're receiving exactly.

loewal, 2022-01-29:
Hello Alexandre,

I am running SPYserver on an Odroid N2+, but also the sdr:// links on https://airspy.com/directory/ are not reacting.
Network is ok, I was a Cisco-teacher ☺️

There is no error whatsoever, because I can’t connect, so even not “play”.

Changing to SDRsharp on my Parallels virtual machine works ok.

It seems, I make a fundamental error somewhere.

Sorry!

AlexandreRouma, 2022-01-29:
you can't enter a sdr:// link, you ahve to manually enter host and port

loewal, 2022-01-29:
Sorry again….

I enter sdr//:10.0.0.60 	next choice  5555
Also tried sdr://10.0.0.60:5555

All this works in SDRsharp….

loewal, 2022-01-29:
> you can't enter a sdr:// link, you have to manually enter host and port

And also 10.0.0.60

Am I completely mistaken perhaps?
Can I listen to websdr’s this way?

loewal, 2022-01-29:
Like this?

AlexandreRouma, 2022-01-29:
like this
[image]

loewal, 2022-01-29:
Sorry, wrong screenshot
[image]

AlexandreRouma, 2022-01-29:
that's the correct way to enter it, check what the spyserver says when trying to connect.

pbij2001, 2022-01-30:
Loewal, Running well here with ddns adress in IP entry. Full IQ restricted to lower sample rates though. SDR++ does not run compressed audio&wfall so needs fast connection.
[image]