AlexandreRouma/SDRPlusPlus #624: Streaming from spyserver stops accidently

Radiomix2000, 2022-01-29:
I have to mention one strange behavior of SDRPP, when connected to spyserver.
The SR setting is 150 000 sps.
Both server and client are in the same LAN.
Connection to server and starting the stream is ok.
SPY Server v2.0.1700
SDRPP v1.0.4
But when you are connected for some time, the stream stops accidently. It may happen even after some seconds after starting stream, but can work 15 minutes without stop the stream. The time when it will happen is impossible to determine. Seems to be random event..

AlexandreRouma, 2022-01-29:
Please show both SDR++ and spyserver logs.
Also what SDR are you using?

Radiomix2000, 2022-01-29:
There is RTL-SDR on server side.
I cannot find spyserver logs. And SDRPP show nothing in terminal when runs. Can you assist?

AlexandreRouma, 2022-01-29:
if you're on windows you have to use the `-s` parameter to enable logging.

Radiomix2000, 2022-01-30:
Strange, but SDRPP doesn't show anything. When streaming stops, nothing happen in log. As well in spyserver.
But I found one interesting and strange thing. The RF gain value influences with stream behavior. I will try to make video to show, what I mean. 
In short - when the gain is less than 15 units, streaming is ok and stable. But when the gain is higher than 18 units, streaming stops shortly, and can start again automatically. When the gain is more than 21 unit, that if stream stops, it stops until I manually disconnect and connect back to server.

Radiomix2000, 2022-01-30:
It looks like so 
https://www.youtube.com/watch?v=X46iG0zdULk

AlexandreRouma, 2022-01-30:
the spyserver logs are what spyserver prints in its window, like SDR++
Can you send those. To me it looks like spyserver's fault right now.

Radiomix2000, 2022-01-30:
The spyserver logs are visible in video on time mark 1:03 min and 2:40 min. They are shown exactly when streaming stops.

AlexandreRouma, 2022-01-30:
Can you try it with SDR# and see if it happens as well?

Radiomix2000, 2022-01-30:
There is no such issue with SDRSharp. With SDRsharp it works flawlessly.

Radiomix2000, 2022-01-31:
Hi Alexandre, if you need more information or need to ask me to make some experimentation, fill free to ask, I'm ready help as much as possible.

AlexandreRouma, 2023-10-30:
Since no other report of this happening has occured in over a year and other apps using my spyserver client code (soapysdr and satdump) haven't had this issue either, I conclude this is a problem with your specific systems (client or server), either at the OS level or the network level.

closed 2023-10-30 by AlexandreRouma