AlexandreRouma/SDRPlusPlus #632: Higher than expected CPU?

TheSkorm, 2022-02-06:
Compiled from master branch this morning and I'm noticing a large amount of CPU usage even when sdrpp isn't connected to an SDR (eg, on first startup). CPU is a `Intel(R) Core(TM) i5-6500T CPU @ 2.50GHz` and on 8GB of ram.

[image]
[image]

Commit: 71c95af7113d90dfdb33da37a7bdcff6e132f380
Ubuntu 21.10 / impish

<details>
  <summary>Log</summary>

```
[2022-02-06 14:04:10.463] [info] SDR++ v1.0.5
[2022-02-06 14:04:10.463] [info] Loading config
[2022-02-06 14:04:10.590] [info] Using OpenGL 3.0
[2022-02-06 14:04:10.700] [info] Loading icons
[2022-02-06 14:04:10.713] [info] Loading band plans
[2022-02-06 14:04:10.719] [info] Loading band plans color table
[2022-02-06 14:04:10.729] [info] Loading modules
[2022-02-06 14:04:10.729] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2022-02-06 14:04:10.734] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2022-02-06 14:04:10.741] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2022-02-06 14:04:10.747] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2022-02-06 14:04:10.754] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2022-02-06 14:04:10.758] [warning] ConfigManager locked, waiting...
[2022-02-06 14:04:10.758] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2022-02-06 14:04:10.763] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-02-06 14:04:10.768] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-02-06 14:04:10.773] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2022-02-06 14:04:10.780] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2022-02-06 14:04:10.788] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-02-06 14:04:10.794] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2022-02-06 14:04:10.799] [info] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[2022-02-06 14:04:10.805] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2022-02-06 14:04:10.813] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2022-02-06 14:04:10.819] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2022-02-06 14:04:10.826] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2022-02-06 14:04:10.832] [info] Initializing Airspy Source (airspy_source)
[2022-02-06 14:04:10.836] [error] Module 'airspy_source' doesn't exist
[2022-02-06 14:04:10.837] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-02-06 14:04:10.841] [error] Module 'airspyhf_source' doesn't exist
[2022-02-06 14:04:10.841] [info] Initializing Audio Sink (audio_sink)
[2022-02-06 14:04:10.845] [info] Initializing BladeRF Source (bladerf_source)
[2022-02-06 14:04:10.848] [error] Module 'bladerf_source' doesn't exist
[2022-02-06 14:04:10.848] [info] Initializing File Source (file_source)
[2022-02-06 14:04:10.853] [info] Initializing Frequency Manager (frequency_manager)
[2022-02-06 14:04:10.860] [info] Initializing HackRF Source (hackrf_source)
[2022-02-06 14:04:10.868] [info] Initializing LimeSDR Source (limesdr_source)
[2022-02-06 14:04:10.873] [error] Module 'limesdr_source' doesn't exist
[2022-02-06 14:04:10.873] [info] Initializing Network Sink (network_sink)
[2022-02-06 14:04:10.877] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-02-06 14:04:10.883] [info] Initializing RFspace Source (rfspace_source)
[2022-02-06 14:04:10.887] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
[2022-02-06 14:04:11.486] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-02-06 14:04:11.492] [info] Initializing Radio (radio)
[2022-02-06 14:04:11.734] [info] FM IF Noise reduction set to 32 taps
[2022-02-06 14:04:11.749] [info] RtAudio stream open
[2022-02-06 14:04:11.749] [info] Initializing Recorder (recorder)
[2022-02-06 14:04:11.772] [info] Initializing Rigctl Server (rigctl_server)
[2022-02-06 14:04:11.776] [info] Initializing SDR++ Server Source (sdrpp_server_source)
[2022-02-06 14:04:11.780] [info] Initializing SDRplay Source (sdrplay_source)
[2022-02-06 14:04:11.784] [error] Module 'sdrplay_source' doesn't exist
[2022-02-06 14:04:11.784] [info] Initializing SoapySDR Source (soapy_source)
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4build1
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
[2022-02-06 14:04:12.680] [info] Setting sample rate to 48000.0
[2022-02-06 14:04:12.680] [info] New DSP samplerate: 48000.0 (source samplerate is 48000.0)
[2022-02-06 14:04:12.681] [info] Updating FFT DSP settings: Keep: 2400, Skip: 0
[2022-02-06 14:04:12.681] [info] Initializing SpyServer Source (spyserver_source)
[2022-02-06 14:04:12.783] [info] Loading color maps
[2022-02-06 14:04:12.798] [info] Loading /usr/share/sdrpp/colormaps/classic_green.json
[2022-02-06 14:04:12.811] [info] Loading /usr/share/sdrpp/colormaps/gqrx.json
[2022-02-06 14:04:12.827] [info] Loading /usr/share/sdrpp/colormaps/turbo.json
[2022-02-06 14:04:12.841] [info] Loading /usr/share/sdrpp/colormaps/classic.json
[2022-02-06 14:04:12.857] [info] Loading /usr/share/sdrpp/colormaps/inferno.json
[2022-02-06 14:04:12.872] [info] Loading /usr/share/sdrpp/colormaps/viridis.json
[2022-02-06 14:04:12.888] [info] Loading /usr/share/sdrpp/colormaps/websdr.json
[2022-02-06 14:04:12.900] [info] Loading /usr/share/sdrpp/colormaps/greyscale.json
[2022-02-06 14:04:12.910] [info] Loading /usr/share/sdrpp/colormaps/magma.json
[2022-02-06 14:04:12.918] [info] Loading /usr/share/sdrpp/colormaps/electric.json
[2022-02-06 14:04:12.925] [info] Loading /usr/share/sdrpp/colormaps/plasma.json
[2022-02-06 14:04:12.950] [info] New DSP samplerate: 2400000.0 (source samplerate is 2400000.0)
[2022-02-06 14:04:12.951] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
[2022-02-06 14:04:12.951] [info] RTLSDRSourceModule 'RTL-SDR Source': Menu Select!
[2022-02-06 14:04:12.951] [info] New DSP samplerate: 2400000.0 (source samplerate is 2400000.0)
[2022-02-06 14:04:12.955] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
[2022-02-06 14:04:13.155] [info] RtAudio stream open
[2022-02-06 14:04:13.180] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
[2022-02-06 14:04:13.180] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
[2022-02-06 14:04:13.183] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
[2022-02-06 14:04:13.188] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 248298652.88841063!
[2022-02-06 14:04:13.188] [info] Running post-init for Audio Sink
[2022-02-06 14:04:13.188] [info] Running post-init for File Source
[2022-02-06 14:04:13.188] [info] Running post-init for Frequency Manager
[2022-02-06 14:04:13.188] [info] Running post-init for HackRF Source
[2022-02-06 14:04:13.188] [info] Running post-init for Network Sink
[2022-02-06 14:04:13.188] [info] Running post-init for PlutoSDR Source
[2022-02-06 14:04:13.188] [info] Running post-init for RFspace Source
[2022-02-06 14:04:13.188] [info] Running post-init for RTL-SDR Source
[2022-02-06 14:04:13.188] [info] Running post-init for RTL-TCP Source
[2022-02-06 14:04:13.188] [info] Running post-init for Radio
[2022-02-06 14:04:13.188] [info] Running post-init for Recorder
[2022-02-06 14:04:13.188] [info] Running post-init for Rigctl Server
[2022-02-06 14:04:13.188] [info] Running post-init for SDR++ Server Source
[2022-02-06 14:04:13.188] [info] Running post-init for SoapySDR Source
[2022-02-06 14:04:13.188] [info] Running post-init for SpyServer Source
[2022-02-06 14:04:13.188] [info] Ready.
```

</details>

darksidelemm, 2022-02-06:
I should note that on the latest M1 build (Built 25th December 2021), I'm seeing 14% CPU use when SDR++ is loaded, not connected to anything.

AlexandreRouma, 2022-02-06:
@TheSkorm you're likely running in software rendered mode, check your intel graphics drivers.

@darksidelemm Remember that on MacOS, just like linux the CPU usage of each process is shown in total percentage. This means that on a 15% usage really is just 1.875% CPU usage.

TheSkorm, 2022-02-06:
yup that would be correct as I'm running this in xrdp and hardware acceleration in that is still work in progress :( https://github.com/neutrinolabs/xrdp/issues/1029

TheSkorm, 2022-02-06:
Went to the effort of setting up the hardware acceleration of xrdp and the CPU usage dropped so this is certainly the cause. not sure if there is anything you can do to reduce the load for software renders ? happy to close this out if you like
[image]

AlexandreRouma, 2022-02-07:
Not possible, this is just the nature of an OpenGL based application.

AlexandreRouma, 2022-02-07:
I'll add that I suggest that if you want to use an SDR over the network, use SDR++ server. It'll do the rendering and processing on the client machine instead.

closed 2022-02-07 by AlexandreRouma

TheSkorm, 2022-02-07:
More so doing it on that machine as it's transmitting and I'm really just using SDR++ as a spy client for rx. Still experimenting with exactly how I want to set things up, so certainly a good idea to use the SDR++ server. I also noticed that resizing the window smaller helped bring CPU load down to an acceptable level too.

Thanks :)