AlexandreRouma/SDRPlusPlus #635: SDRPlusPlus fails to build on Linux Mint

mmarsden191, 2022-02-07:
I am running Linux Mint 20.3 and have downloaded the latest SDRPlusPlus targall source from the releases page.  But when I try to build the project, many errors are produced.  Any help would be appreciated.  Here is a log of the failed build process.
mjm@dv8-PC:/media/mjm/8C4A03E34A03C942/SDR++/SDRPlusPlus-1.0.4/build$ make -j8
[  0%] Built target error_sim
[  1%] Built target correct-sse-h
[  1%] Built target error_sim_sse
[  1%] Built target correct-h
[  3%] Built target error_sim_shim
[  7%] Built target correct-reed-solomon
[ 13%] Built target correct-convolutional
[ 16%] Built target correct-convolutional-sse
[ 17%] Built target correct_static
[ 22%] Built target correct
[ 23%] Built target discord-rpc
[ 65%] Built target sdrpp_core
[ 66%] Linking CXX executable sdrpp
[ 67%] Built target airspy_source
[ 70%] Built target file_source
[ 71%] Built target rtl_sdr_source
[ 73%] Built target airspyhf_source
[ 73%] Built target soapy_source
[ 75%] Built target hackrf_source
[ 76%] Built target rtl_tcp_source
[ 78%] Built target spyserver_source
[ 79%] Built target network_sink
[ 83%] Built target plutosdr_source
[ 84%] Built target audio_sink
Scanning dependencies of target frequency_manager
Scanning dependencies of target discord_integration
[ 86%] Built target radio
[ 88%] Built target meteor_demodulator
[ 90%] Built target weather_sat_decoder
Scanning dependencies of target rigctl_server
Scanning dependencies of target recorder
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `void fmt::v6::internal::buffer<char>::append<char>(char const*, char const*) [clone .constprop.0]':
main.cpp:(.text+0x181): undefined reference to `operator new(unsigned long)'
/usr/bin/ld: main.cpp:(.text+0x1ab): undefined reference to `operator delete(void*)'
/usr/bin/ld: main.cpp:(.text+0x230): undefined reference to `std::__throw_bad_alloc()'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `fmt::v6::basic_memory_buffer<char, 500ul, std::allocator<char> >::~basic_memory_buffer()':
main.cpp:(.text._ZN3fmt2v619basic_memory_bufferIcLm500ESaIcEED2Ev[_ZN3fmt2v619basic_memory_bufferIcLm500ESaIcEED5Ev]+0x20): undefined reference to `operator delete(void*)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `fmt::v6::basic_memory_buffer<char, 250ul, std::allocator<char> >::~basic_memory_buffer()':
main.cpp:(.text._ZN3fmt2v619basic_memory_bufferIcLm250ESaIcEED2Ev[_ZN3fmt2v619basic_memory_bufferIcLm250ESaIcEED5Ev]+0x20): undefined reference to `operator delete(void*)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `spdlog::details::full_formatter::~full_formatter()':
main.cpp:(.text._ZN6spdlog7details14full_formatterD2Ev[_ZN6spdlog7details14full_formatterD5Ev]+0x2b): undefined reference to `operator delete(void*)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `fmt::v6::basic_memory_buffer<unsigned int, 32ul, std::allocator<unsigned int> >::~basic_memory_buffer()':
main.cpp:(.text._ZN3fmt2v619basic_memory_bufferIjLm32ESaIjEED2Ev[_ZN3fmt2v619basic_memory_bufferIjLm32ESaIjEED5Ev]+0x20): undefined reference to `operator delete(void*)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `spdlog::details::aggregate_formatter::~aggregate_formatter()':
main.cpp:(.text._ZN6spdlog7details19aggregate_formatterD2Ev[_ZN6spdlog7details19aggregate_formatterD5Ev]+0x20): undefined reference to `operator delete(void*)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o:main.cpp:(.text._ZNSt23_Sp_counted_ptr_inplaceIN6spdlog6loggerESaIS1_ELN9__gnu_cxx12_Lock_policyE2EE10_M_destroyEv[_ZNSt23_Sp_counted_ptr_inplaceIN6spdlog6loggerESaIS1_ELN9__gnu_cxx12_Lock_policyE2EE10_M_destroyEv]+0x5): more undefined references to `operator delete(void*)' follow
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `spdlog::details::color_stop_formatter::~color_stop_formatter()':
main.cpp:(.text._ZN6spdlog7details20color_stop_formatterD0Ev[_ZN6spdlog7details20color_stop_formatterD5Ev]+0xa): undefined reference to `operator delete(void*, unsigned long)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `spdlog::details::color_start_formatter::~color_start_formatter()':
main.cpp:(.text._ZN6spdlog7details21color_start_formatterD0Ev[_ZN6spdlog7details21color_start_formatterD5Ev]+0xa): undefined reference to `operator delete(void*, unsigned long)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `spdlog::details::ch_formatter::~ch_formatter()':
main.cpp:(.text._ZN6spdlog7details12ch_formatterD0Ev[_ZN6spdlog7details12ch_formatterD5Ev]+0xa): undefined reference to `operator delete(void*, unsigned long)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `spdlog::details::elapsed_formatter<spdlog::details::null_scoped_padder, std::chrono::duration<long, std::ratio<1l, 1l> > >::~elapsed_formatter()':
main.cpp:(.text._ZN6spdlog7details17elapsed_formatterINS0_18null_scoped_padderENSt6chrono8durationIlSt5ratioILl1ELl1EEEEED0Ev[_ZN6spdlog7details17elapsed_formatterINS0_18null_scoped_padderENSt6chrono8durationIlSt5ratioILl1ELl1EEEEED5Ev]+0xa): undefined reference to `operator delete(void*, unsigned long)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `spdlog::details::elapsed_formatter<spdlog::details::null_scoped_padder, std::chrono::duration<long, std::ratio<1l, 1000l> > >::~elapsed_formatter()':
main.cpp:(.text._ZN6spdlog7details17elapsed_formatterINS0_18null_scoped_padderENSt6chrono8durationIlSt5ratioILl1ELl1000EEEEED0Ev[_ZN6spdlog7details17elapsed_formatterINS0_18null_scoped_padderENSt6chrono8durationIlSt5ratioILl1ELl1000EEEEED5Ev]+0xa): undefined reference to `operator delete(void*, unsigned long)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o:main.cpp:(.text._ZN6spdlog7details17elapsed_formatterINS0_18null_scoped_padderENSt6chrono8durationIlSt5ratioILl1ELl1000000EEEEED0Ev[_ZN6spdlog7details17elapsed_formatterINS0_18null_scoped_padderENSt6chrono8durationIlSt5ratioILl1ELl1000000EEEEED5Ev]+0xa): more undefined references to `operator delete(void*, unsigned long)' follow
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `fmt::v6::basic_memory_buffer<char, 500ul, std::allocator<char> >::~basic_memory_buffer()':
main.cpp:(.text._ZN3fmt2v619basic_memory_bufferIcLm500ESaIcEED0Ev[_ZN3fmt2v619basic_memory_bufferIcLm500ESaIcEED5Ev]+0x20): undefined reference to `operator delete(void*)'
/usr/bin/ld: main.cpp:(.text._ZN3fmt2v619basic_memory_bufferIcLm500ESaIcEED0Ev[_ZN3fmt2v619basic_memory_bufferIcLm500ESaIcEED5Ev]+0x2e): undefined reference to `operator delete(void*, unsigned long)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `fmt::v6::basic_memory_buffer<char, 250ul, std::allocator<char> >::~basic_memory_buffer()':
main.cpp:(.text._ZN3fmt2v619basic_memory_bufferIcLm250ESaIcEED0Ev[_ZN3fmt2v619basic_memory_bufferIcLm250ESaIcEED5Ev]+0x20): undefined reference to `operator delete(void*)'
/usr/bin/ld: main.cpp:(.text._ZN3fmt2v619basic_memory_bufferIcLm250ESaIcEED0Ev[_ZN3fmt2v619basic_memory_bufferIcLm250ESaIcEED5Ev]+0x2e): undefined reference to `operator delete(void*, unsigned long)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `spdlog::details::full_formatter::~full_formatter()':
main.cpp:(.text._ZN6spdlog7details14full_formatterD0Ev[_ZN6spdlog7details14full_formatterD5Ev]+0x2b): undefined reference to `operator delete(void*)'
/usr/bin/ld: main.cpp:(.text._ZN6spdlog7details14full_formatterD0Ev[_ZN6spdlog7details14full_formatterD5Ev]+0x39): undefined reference to `operator delete(void*, unsigned long)'
/usr/bin/ld: CMakeFiles/sdrpp.dir/src/main.cpp.o: in function `fmt::v6::basic_memory_buffer<unsigned int, 32ul, std::allocator<unsigned int> >::~basic_memory_buffer()':
main.cpp:(.text._ZN3fmt2v619basic_memory_bufferIjLm32ESaIjEED0Ev[_ZN3fmt2v619basic_memory_bufferIjLm32ESaIjEED5Ev]+0x20): undefined reference to `operator delete(void*)'
[... 705 lines cut ...]
                 from /usr/include/c++/9/ext/string_conversions.h:43,
                 from /usr/include/c++/9/bits/basic_string.h:6493,
                 from /usr/include/c++/9/string:55,
                 from /usr/include/c++/9/stdexcept:39,
                 from /usr/include/c++/9/array:39,
                 from /usr/include/c++/9/tuple:39,
                 from /usr/include/c++/9/bits/unique_ptr.h:37,
                 from /usr/include/c++/9/memory:80,
                 from /media/mjm/8C4A03E34A03C942/SDR++/SDRPlusPlus-1.0.4/core/src/spdlog/common.h:12,
[... cut ...]
      |          ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   37 |       __bos (__s), __fmt, __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 96%] Linking CXX shared library discord_integration.so
[ 96%] Built target discord_integration
[ 97%] Linking CXX shared library rigctl_server.so
[ 97%] Built target rigctl_server
[ 97%] Linking CXX shared library recorder.so
[ 97%] Built target recorder
[ 98%] Linking CXX shared library frequency_manager.so
[ 98%] Built target frequency_manager
make: *** [Makefile:130: all] Error 2

AlexandreRouma, 2022-02-08:
Ok so I don't know what's going on with the compiler error, probably something wrong with your environment or broken compiler install.

In any case, using the debian bullseye build isn't garanteed to work. You have to download the Ubuntu Focal deb.

mmarsden191, 2022-02-09:
Whenever I tried building the project from the command line in a terminal, the source files seemed to compile properly but there were failures during the linking process, as if I was missing a library.  So, I did a "make clean".  Then, I deleted the build folder, and did a "cmake ." in the top-level project folder. Next, I opened the project in Eclipse CDT and it compiled and linked successfully.  Thanks for your help.

closed 2022-02-10 by AlexandreRouma