AlexandreRouma/SDRPlusPlus #636: Still can't get SDR++ to run

w9ran, 2022-02-08:
I've removed old redistributables and installed this one which according to Microsoft is supposed to be compatible with everything back to 2015 to present.   But after reboot SDR++ still won't run, even though it no longer complains about the missing DLL.    Now all that happens is a window flashes for an instant, not long enough to see anything but a brief flicker.   Same when using Run as Administrator.    This is with a freshly downloaded copy of v 1.0.4  Win 10 x64, updated.

Is there any way to run in a verbose mode or to create a logfile or something to try to at least see what's going on?   Any other suggestions would be appreciated.  
Thanks,
Bob

shuyuan-liu, 2022-02-08:
> verbose mode

Run sdrpp from a terminal with the `-s` option.

w9ran, 2022-02-08:
Ah...thanks:

[2022-02-08 09:32:09.304] [error] Glfw Error 65542: WGL: The driver does not appear to support OpenGL
[2022-02-08 09:32:09.305] [info] OpenGL 2.1 was not supported

So it looks like I'm done on this machine at least.   It's a Dell i7-2600 with 16GB but evidently insufficient video resources.
Bob

w9ran, 2022-02-08:
Thanks for clarifying. I do have known video problems on this PC after a Windows update broke something.  The best driver I can get to install doesn't support openGL above v1.1 per OpenGL Extensions viewer).   It's  a Dell Optiplex 990 i7-2600 with 16GB but in Dell's opinion, that's insufficient to support Windows 10 so I cannot install the driver that might fix this problem.  And Microsoft says it doesn't meet their minimum requirements for Windows 11.    So I think my only alternative if I want to run SDR++ is to either add a new video card where hopefully the drivers will support the required level of openGL or buy a new PC (which is obviously what they prefer!)

73, Bob W9RAN

AlexandreRouma, 2022-02-08:
The i7-2600 has an HD2000 integrated GPU which supports OpenGL 3.1
That is way more than enough for windows 10 or 11 (but avoid 11 anyway).
Just download the drivers from Intel's website instead of Dell.

BT-Justice, 2022-02-11:
@w9ran
Intel has the Intel® Driver & Support Assistant (Intel® DSA) software which will find the latest Intel drivers for your computer.

https://www.intel.com/content/www/us/en/support/detect.html

AND... make some new YouTube SDR videos please!

AlexandreRouma, 2022-09-13:
closing as idle, if still not fixed with the latest nightly please reopen

closed 2022-09-13 by AlexandreRouma