AlexandreRouma/SDRPlusPlus #644: 64-bit Raspberry Pi OS (28 Jan '22 version, Bullseye): Anyone successfully install SDR++ ?

joaquinhar, 2022-02-11:
I made a fresh SD card with the recently-released Raspberry Pi 64-bit OS and used the rpi_install.sh to build from src.
It fails with this error from the CMakeError.log file:

/usr/bin/cc  -fPIC -std=c99 -Wall -Wpedantic -O2 -DCHECK_FUNCTION_EXISTS=dotprod -rdynamic CMakeFiles/c
mTC_e134d.dir/CheckFunctionExists.c.o -o cmTC_e134d  -lFEC 
/usr/bin/ld: cannot find -lFEC

Apologies....I'm probably in way over my head.

RyLeeRyno, 2022-02-12:
you don't need to run that script.  just install the following dependencies and built from the git folder.

sudo apt update
sudo apt install -y build-essential cmake git libfftw3-dev libglfw3-dev libvolk2-dev libzstd-dev librtaudio-dev p7zip-full wget

cd build
cmake ../
make -j4
sudo make install

my system.
pi@RPi4-v3:~/PiSDR/Software/SDRPlusPlus $ uname -a
Linux RPi4-v3 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux
pi@RPi4-v3:~/PiSDR/Software/SDRPlusPlus $ sdrpp -a
[2022-02-11 21:56:07.624] [info] SDR++ v1.0.5

joaquinhar, 2022-02-12:
> 

Thanks very much, RyLeeRyno.  
Your changes made the process better.  However, this RPi 3B+ hangs during the **make**.  It is a "complete hang", even the desktop clock doesn't change.  So, no errors now, but no completed **make** either. Eventually, the green light just goes into a steady mode and after about 10 minutes, the monitor has no signal. CTRL-ALT-DEL doesn't even stop it. 
 
I'm guessing this is an issue with the RAM of the RPi 3B+ which has had issues with the 32-bit version of the OS where it would hang with just a few simple web pages open in Chromium.

If anyone has any other suggestions and has made this build work on a 3B+, I am very interested.  Otherwise, I may go back to the 32-bit OS and keep using gqrx

RyLeeRyno, 2022-02-13:
delete the build folder and run the cmake again with the below options.

cmake .. -DOPT_BUILD_LIMESDR_SOURCE=OFF -DOPT_BUILD_AIRSPYHF_SOURCE=OFF DOPT_BUILD_BLADERF_SOURCE=OFF -DOPT_BUILD_FILE_SOURCE=OFF -DOPT_BUILD_HACKRF_SOURCE=OFF -DOPT_BUILD_RFSPACE_SOURCE=OFF -DOPT_BUILD_RTL_TCP_SOURCE=OFF -DOPT_BUILD_SPYSERVER_SOURCE=OFF -DOPT_BUILD_PLUTOSDR_SOURCE=OFF -DOPT_BUILD_METEOR_DEMODULATOR=OFF -DOPT_BUILD_DISCORD_PRESENCE=OFF -DOPT_BUILD_RIGCTL_SERVER=OFF

make -j4

AlexandreRouma, 2022-02-13:
Why in the world would you do that.... If you're gonna disable all those features you might as well use GQRX...

joaquinhar, 2022-02-13:
> delete the build folder and run the cmake again with the below options.
>
> cmake .. -DOPT_BUILD_LIMESDR_SOURCE=OFF -DOPT_BUILD_AIRSPYHF_SOURCE=OFF DOPT_BUILD_BLADERF_SOURCE=OFF [...]

Thanks again.  
The RPi 3B+ still hangs.
Looks like I'll switch to an old  Windows laptop and try to build sdrpp on that.

RyLeeRyno, 2022-02-14:
I suggested minimizing the build for a smaller footprint since there was a concern that the Pi3+ low memory and arm64 architecture played a part.

If successful then of course rebuild with the defaults  enabled x camemory install with the intent of adding the packages back if the install was successful.  

-Ryan

AlexandreRouma, 2022-02-16:
you're not gonna gain that much ram by disabling all of those and making the install basically useless

SanchYESS, 2022-02-20:
Got total freezes on RPi 4 while make -j4, -j3 resolved the issue

toxa1983poloskov, 2022-02-22:
use only the parameter:   make 
then pi  on  1GB RAM does not hang

toxa1983poloskov, 2022-02-22:
64-bit Raspberry Pi OS  (Bullseye)

tell me what the problem is RtAudio ?    the program starts and closes.

terminate called after throwing an instance of 'RtAudioError'   
  what():  RtApiPulse::stopStream(): the stream is not open!
Аварийный останов

how do I adjust the sound?
[rtaudio]  (https://github.com/thestk/rtaudio)  I have collected

AlexandreRouma, 2022-09-13:
For future reference, building on RaspiOS bullseye 64bit is exactly the same as on a desktop debian bullseye.

closed 2022-09-13 by AlexandreRouma