AlexandreRouma/SDRPlusPlus #646: Hello, I have (HackRF ) I want to open the code of the signals (VHF ) and (UHF ) on the FM channel on the laptop using Windows 7, Thanks

abbas1964, 2022-02-11:
Hello, I have (HackRF ) I want to open the code of the signals (VHF ) and (UHF ) on the FM channel on the laptop using Windows 7, Thanks

shuyuan-liu, 2022-02-11:
OK, what exactly do you need help with? What have you tried? Also, you opened two Issues with the same title, please close one of them.

AlexandreRouma, 2022-02-11:
I gotta say I'm really not sure what's being asked here. If it's **not** related to the SDR++ software I suggest asking on a general SDR related forum instead.

abbas1964, 2022-02-11:
There are no two issues, but only one is that I have SDRSharp on my laptop and I have a HackRF device, NooElec............ and I can't decode the signal sent via FM

AlexandreRouma, 2022-02-11:
This is not the support section for SDRsharp but for SDR++.
Please ask for support on Airspy's forums instead.

closed 2022-02-11 by AlexandreRouma

abbas1964, 2022-02-22:
> 

Good
I need the latest DSD software

abbas1964, 2022-02-22:
> يجب أن أقول إنني حقًا لست متأكدًا مما يُطلب هنا. إذا **لم يكن** مرتبطًا ببرنامج SDR ++ ، فأقترح أن تسأل في منتدى عام متعلق بحقوق السحب الخاصة بدلاً من ذلك.

Good
I need the latest DSD software

abbas1964, 2022-02-22:
> حسنًا ، ما الذي تحتاج إلى المساعدة فيه بالضبط؟ ماذا حاولت؟ بدون مزيد من التفاصيل ، لن نعرف المشكلة التي تواجهها ولن تتمكن من مساعدتك.
> 
> يبدو أن فتحت ثلاث إصدارات الإصدارات نفسها ، هل يمكنك إغلاق الإصدارات الأخرى من فضلك؟ يؤدي إلى إقرأ النكسات.
> 
> يرجى ملاحظة أن SDR ++ تم تطويره الأشخاص في أوقات فراغهم ، ولا يعمل أحد بدوام كامل للإجابة على الأسئلة على هذا الموقع. سيكون صبرك موضع تقدير كبير. شكرا لك!
Good
I need the latest DSD software