AlexandreRouma/SDRPlusPlus #650: Close window not possible

loewal, 2022-02-14:
Bonjour Edouard,

I installed the latest version under Ubuntu Impish.
Now I cannot close the window anymore and have to kill the process within  BTOP (or any other processsmanager).
This happens on 2 machines.
On my iMac this is not the case.
Can you help?

AlexandreRouma, 2022-02-14:
What do you call "latest version"? This doesn't tell me anything about which build you're actually using.
Latest released version? Latest master branch build? latest backend_abstraction branch build?

If you're talking about the backend_abstraction branch, there's a reason it hasn't been merged into master yet. It still has issues and this exact one was fixed this morning but not pushed yet.

loewal, 2022-02-14:
Hello Alexandre,

I downloaded the version under Actions/ “backend_abstraction” and then the Impish-file.

https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/1837378817

AlexandreRouma, 2022-02-16:
The non master branches should not be used. they're work in progress and usually broken.
Only install builds of the master branch

closed 2022-02-16 by AlexandreRouma