AlexandreRouma/SDRPlusPlus #652: SDR++ Looks Blurry On Mac M1 and Help For M1 Mac

ChrisDev8, 2022-02-16:
When I was trying to build my own version of SDR++, the package volk would not install and I got an error. I don't remember what it said and something happened to my homebrew install so now I can't use brew. But the github issue #557 had the zip file with SDR++.app that worked perfectly. So to the creator, please add that zip file on issue #557 to the releases page. Anyways, once I got it to work it looked a little blurry on apple retina screen and a little off. I hope you understand!!!! Please don't be mean I've had lots of people who like NO WAY THAT A DUMB IDEA JUST CLOSE IT GDFGTERTWRGRGRE.

[image]

AlexandreRouma, 2022-02-17:
Can't do anything about it. SDR++ renders through OpenGL, if apple applies sont dumb scaling to stuff I can't do anything about it.

ChrisDev8, 2022-02-17:
Ok but how about adding a `SDR++.app` file to the release page. Just like the .app file in issue #557 so people don't have to build their own. Not many people are "Computer Geeks".

AlexandreRouma, 2022-02-17:
It's not the same version. It was a special manual build of the master branch, I'm not adding it to an existing release.
When the time of releasing the next version comes around, it will be in there, not before.

closed 2022-02-17 by ChrisDev8