AlexandreRouma/SDRPlusPlus #656: MAC issue zsh: segmentation fault  sdrpp

V0YNIC, 2022-02-17:
With RTL-SDR, crashes when hitting play 

[2022-02-17 22:00:35.623] [info] SDR++ v1.0.4
[2022-02-17 22:00:35.623] [info] Loading config
[2022-02-17 22:00:37.161] [info] Loading icons
[2022-02-17 22:00:37.174] [info] Loading band plans
[2022-02-17 22:00:37.189] [info] Loading band plans color table
[2022-02-17 22:00:37.212] [info] Loading modules
[2022-02-17 22:00:37.212] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_sdr_source.dylib
[2022-02-17 22:00:37.227] [info] Loading /usr/local/lib/sdrpp/plugins/meteor_demodulator.dylib
[2022-02-17 22:00:37.244] [info] Loading /usr/local/lib/sdrpp/plugins/airspy_source.dylib
[2022-02-17 22:00:37.260] [info] Loading /usr/local/lib/sdrpp/plugins/network_sink.dylib
[2022-02-17 22:00:37.277] [info] Loading /usr/local/lib/sdrpp/plugins/radio.dylib
[2022-02-17 22:00:37.293] [info] Loading /usr/local/lib/sdrpp/plugins/rigctl_server.dylib
[2022-02-17 22:00:37.311] [info] Loading /usr/local/lib/sdrpp/plugins/audio_sink.dylib
[2022-02-17 22:00:37.326] [info] Loading /usr/local/lib/sdrpp/plugins/m17_decoder.dylib
[2022-02-17 22:00:37.341] [info] Loading /usr/local/lib/sdrpp/plugins/frequency_manager.dylib
[2022-02-17 22:00:37.362] [info] Loading /usr/local/lib/sdrpp/plugins/recorder.dylib
[2022-02-17 22:00:37.378] [info] Loading /usr/local/lib/sdrpp/plugins/file_source.dylib
[2022-02-17 22:00:37.395] [info] Loading /usr/local/lib/sdrpp/plugins/airspyhf_source.dylib
[2022-02-17 22:00:37.411] [info] Loading /usr/local/lib/sdrpp/plugins/spyserver_source.dylib
[2022-02-17 22:00:37.427] [info] Loading /usr/local/lib/sdrpp/plugins/new_portaudio_sink.dylib
[2022-02-17 22:00:37.445] [info] Loading /usr/local/lib/sdrpp/plugins/discord_integration.dylib
[2022-02-17 22:00:37.459] [info] Loading /usr/local/lib/sdrpp/plugins/weather_sat_decoder.dylib
[2022-02-17 22:00:37.475] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_tcp_source.dylib
[2022-02-17 22:00:37.495] [info] Loading /usr/local/lib/sdrpp/plugins/hackrf_source.dylib
[2022-02-17 22:00:37.510] [info] Initializing Airspy Source (airspy_source)
[2022-02-17 22:00:37.537] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-02-17 22:00:37.548] [info] Initializing Audio Sink (audio_sink)
[2022-02-17 22:00:37.609] [info] Initializing BladeRF Source (bladerf_source)
[2022-02-17 22:00:37.610] [error] Module 'bladerf_source' doesn't exist
[2022-02-17 22:00:37.610] [info] Initializing File Source (file_source)
[2022-02-17 22:00:37.623] [info] Initializing Frequency Manager (frequency_manager)
[2022-02-17 22:00:37.639] [info] Initializing HackRF Source (hackrf_source)
[2022-02-17 22:00:37.665] [info] Initializing LimeSDR Source (limesdr_source)
[2022-02-17 22:00:37.673] [error] Module 'limesdr_source' doesn't exist
[2022-02-17 22:00:37.673] [info] Initializing Network Sink (network_sink)
[2022-02-17 22:00:37.690] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-02-17 22:00:37.705] [error] Module 'plutosdr_source' doesn't exist
[2022-02-17 22:00:37.705] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Found Rafael Micro R820T tuner
[2022-02-17 22:00:38.207] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-02-17 22:00:38.208] [info] Initializing Radio (radio)
[2022-02-17 22:00:38.468] [info] Audio device open.
[2022-02-17 22:00:38.468] [info] Initializing Recorder (recorder)
[2022-02-17 22:00:38.470] [info] Initializing Rigctl Server (rigctl_server)
[2022-02-17 22:00:38.488] [info] Initializing SDRplay Source (sdrplay_source)
[2022-02-17 22:00:38.505] [error] Module 'sdrplay_source' doesn't exist
[2022-02-17 22:00:38.505] [info] Initializing SoapySDR Source (soapy_source)
[2022-02-17 22:00:38.521] [error] Module 'soapy_source' doesn't exist
[2022-02-17 22:00:38.521] [info] Initializing SpyServer Source (spyserver_source)
[2022-02-17 22:00:38.555] [info] Loading color maps
[2022-02-17 22:00:38.571] [info] Loading /usr/local/share/sdrpp/colormaps/classic.json
[2022-02-17 22:00:38.588] [info] Loading /usr/local/share/sdrpp/colormaps/inferno.json
[2022-02-17 22:00:38.604] [info] Loading /usr/local/share/sdrpp/colormaps/turbo.json
[2022-02-17 22:00:38.621] [info] Loading /usr/local/share/sdrpp/colormaps/gqrx.json
[2022-02-17 22:00:38.638] [info] Loading /usr/local/share/sdrpp/colormaps/electric.json
[2022-02-17 22:00:38.655] [info] Loading /usr/local/share/sdrpp/colormaps/plasma.json
[2022-02-17 22:00:38.672] [info] Loading /usr/local/share/sdrpp/colormaps/greyscale.json
[2022-02-17 22:00:38.688] [info] Loading /usr/local/share/sdrpp/colormaps/viridis.json
[2022-02-17 22:00:38.705] [info] Loading /usr/local/share/sdrpp/colormaps/magma.json
[2022-02-17 22:00:38.722] [info] Loading /usr/local/share/sdrpp/colormaps/websdr.json
[2022-02-17 22:00:38.738] [info] Loading /usr/local/share/sdrpp/colormaps/classic_green.json
[2022-02-17 22:00:38.752] [info] New DSP samplerate: 2400000.0 (source samplerate is 2400000.0)
[2022-02-17 22:00:38.752] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
[2022-02-17 22:00:38.752] [info] RTLSDRSourceModule 'RTL-SDR Source': Menu Select!
[2022-02-17 22:00:38.752] [info] New DSP samplerate: 2400000.0 (source samplerate is 2400000.0)
[2022-02-17 22:00:38.752] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
[2022-02-17 22:00:39.071] [info] Audio device open.
[2022-02-17 22:00:39.087] [info] Updating FFT DSP settings: Keep: 65536, Skip: 54464
[2022-02-17 22:00:39.087] [info] Updating FFT DSP settings: Keep: 65536, Skip: 105892
[2022-02-17 22:00:39.089] [info] Updating FFT DSP settings: Keep: 65536, Skip: 105892
[2022-02-17 22:00:39.090] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 90981012.05859375!
[2022-02-17 22:00:39.090] [info] Running post-init for Airspy Source
[2022-02-17 22:00:39.090] [info] Running post-init for AirspyHF+ Source
[2022-02-17 22:00:39.090] [info] Running post-init for Audio Sink
[2022-02-17 22:00:39.090] [info] Running post-init for File Source
[2022-02-17 22:00:39.090] [info] Running post-init for Frequency Manager
[2022-02-17 22:00:39.090] [info] Running post-init for HackRF Source
[2022-02-17 22:00:39.090] [info] Running post-init for Network Sink
[2022-02-17 22:00:39.090] [info] Running post-init for RTL-SDR Source
[2022-02-17 22:00:39.090] [info] Running post-init for RTL-TCP Source
[2022-02-17 22:00:39.090] [info] Running post-init for Radio
[2022-02-17 22:00:39.090] [info] Running post-init for Recorder
[2022-02-17 22:00:39.090] [info] Running post-init for Rigctl Server
[2022-02-17 22:00:39.090] [info] Running post-init for SpyServer Source
[2022-02-17 22:00:39.090] [info] Ready.
Found Rafael Micro R820T tuner
[2022-02-17 22:00:41.686] [info] RTL-SDR Sample Rate: 2400000.0
[R82XX] PLL not locked!
Disabled direct sampling mode
[2022-02-17 22:00:42.259] [info] RTLSDRSourceModule 'RTL-SDR Source': Start!
[2022-02-17 22:00:42.315] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 90981012.05859375!
zsh: segmentation fault  sdrpp

V0YNIC, 2022-02-17:
8758 Segmentation fault: 11  sdrpp

AlexandreRouma, 2022-02-17:
1.0.4 is getting old, use https://www.sdrpp.org/nightly

closed 2022-02-17 by V0YNIC