AlexandreRouma/SDRPlusPlus #657: Correct way to install on macOs

60mirrors, 2022-02-18:
Bonjour Alexandre,

I am confused:

I compiled SDR++ with support for SoapySDR and SDRplay and have to run from /usr/local/bin. Works fine!
Then I thought that a simple download of the app for macOs would be the thing to do in the future.
But now I have a version with ánd without support on different locations..

Do I have to compile from now on?
By the way: running from the commandrompt doesn't give me that nice app-icon.

Can you help?

AlexandreRouma, 2022-02-18:
Hello,
I don't recommend building yourself on MacOS, it's not yet well supported or easy.

Download the prebuilt .app bundle at https://www.sdrpp.org/nightly and install the SDRplay API.
It's for intel only but you haven't mentioned which CPU architecture or OSX version your mac uses.

60mirrors, 2022-02-18:
Hello Alexandre,

This is the output of SDRUtil —info:

/Users/Loewal/Downloads/ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so (0.3.2)
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-1c6631d)
Module found: /usr/local/lib/SoapySDR/modules0.8/libuhdSupport.so (0.4.1-1f7b6fa)
Available factories... rtlsdr, sdrplay, uhd
Available converters...
 - CF32 -> [CF32, CS16, CS8, CU16, CU8]
 - CS16 -> [CF32, CS16, CS8, CU16, CU8]
 - CS32 -> [CS32]
 - CS8 -> [CF32, CS16, CS8, CU16, CU8]
 - CU16 -> [CF32, CS16, CS8]
 - CU8 -> [CF32, CS16, CS8]
 - F32 -> [F32, S16, S8, U16, U8]
 - S16 -> [F32, S16, S8, U16, U8]
 - S32 -> [S32]
 - S8 -> [F32, S16, S8, U16, U8]
 - U16 -> [F32, S16, S8]
 - U8 -> [F32, S16, S8]
***@***.***█▓▒░ vr feb 18 08:38:36P
/Users/Loewal/Downloads/

Other programs do show my RSP1, directly and through Soapy.

By the way: compiling isn’t that difficult and works well.
The only “problem” is, that the way I start SDR++ from the prompt isn’t that nice as starting the normal way.

How do I get more than Airpyhf+, Hackrf and RTLSDR?

Thank you.

AlexandreRouma, 2022-02-18:
Firstly, there's a reason I'm telling you to download the prebuilt one instead of building it yourself. It WILL NOT create an icon because that is NOT how apps work on MacOS. It has to be built in a specific way and since it's a work in progress I don't support it publically right now. Download the prebuild bundle and remove the one you built.

Secondly, don't use SoapySDR for the SDRplay, SDR++ has **native support** for SDRplay SDRs and almost all other SDRs on the market, soapy support is just there for people running esoteric hardware. I will stop supporting SoapySDR once all SDRs soapy provides support for are natively supported.

Finally, you still haven't told me what CPU architecture you're using, can't help you with the SDRplay stuff if you don't.....

AlexandreRouma, 2022-02-18:
Install the latest version of the SDRplay API for MacOS and make sure it's running. it's not shipped with SDR++

gvanem, 2022-02-18:
@loewal This looks wrong (your Soapy-SDR is messed up?):

>  `hardware=B0020P0004`
> `sdrplay_api_api_version=0.000000`

Here it reports:
``` 
 driver=SDRplay
 hardware=RSP1A
 mir_sdr_api_version=2.130000
 mir_sdr_hw_version=255
 serial=1803049E94
```
So you'll need to update it.

loewal, 2022-02-18:
Hello Gisle,

I also see an other difference: mir_sdr_api_version etc….
Mine is sdrplay….
Where did you get your api?

loewal, 2022-02-18:
I downloaded and installed form the SDRPlay-site.

For: RSP1
OS: MacOS other

So… what else can I do?

AlexandreRouma, 2022-02-18:
https://www.sdrplay.com/softwarehome/

loewal, 2022-02-18:
That’s the one I installed (again) this morning.
Even restarted my iMac

????

n2iw, 2022-02-20:
> 

As Alex said, don't try to use Soapy to do anything SDRPlay related. In my setup, Soapy can't find SDRPlay at all, but it works fine in SDR++. 

If you don't see SDRPlay in your sources list, try to quit SDR++ then delete "/Users/<your-user-name>/Library/Application Support/sdrpp", then restart SDR++. I recently lost SDRPlay in the sources list, it only appeared after I delete that folder.

nechry, 2022-07-22:
I download the prebuilt .app bundle for OSX at https://www.sdrpp.org/nightly as recommended, but I expect to use a ADALM-PLUTO (PlutoSDR). But in build option the PlutoSDR is turn OFF. Is something I can do to get it work at the moment? thank-you

AlexandreRouma, 2022-07-23:
I had to stop including the plutosdr support in MacOS because the driver was broken. I'll retry again later

nechry, 2022-07-23:
ok thanks for your reply, anyway I also have a rtl-sdr stick and I really appreciate your development, is my favorite srd tool so far