AlexandreRouma/SDRPlusPlus #660: Visual glitches (ghosting) on Linux Mint 20.3 / AMD Ryzen 4300U

SanchYESS, 2022-02-20:
Hello! I've got weird behavior on Mint 20.3 kernel 5.13.0-22-generic.
Seems like ghosting, like image is accumulated instead of updating. SDR++ is working, I can hear tuned stations. FFT is working, waterfall is absent, menu and top bar are totally glitchy but also working.
SDR++ is built from source.
Runs on Lenovo IdeaPad Flex 5 14ARE05, AMD Ryzen 4300U.
[image]

AlexandreRouma, 2022-02-20:
Broken graphics driver, reinstall threm.

closed 2022-02-20 by AlexandreRouma

toxa1983poloskov, 2022-02-22:
> Broken graphics driver, reinstall them.
thanks!
the same problem is observed on the old GNU NVIDIA Corporation G86 [GeForce 8400 GS] c Nvidia binary driver 340.108.

xserver-xorg-video-nouveau --  the graphical interface is usually displayed!!