AlexandreRouma/SDRPlusPlus #663: Feature request

3Domse3, 2022-02-20:
- recording audio and baseband simultaneously
- baseband replay controls for scrubbing and pausing and recording info view with length, frequency etc

AlexandreRouma, 2022-02-20:
1) Just add a second recorder instance and start both (Module Manager -> select "recorder" in the drop down at the bottom -> give it a name like "Recorder 2" in the text field next to the drop down -> press "+"
2) Already requested and planned

AlexandreRouma, 2022-02-20:
whoops wrong button, wil close once the n°2 is available

3Domse3, 2022-02-21:
Oh wow! I didn't know SDR++ was so customizable :D Thanks for the tip!

3Domse3, 2022-02-22:
Some more ideas:
- make bookmarks editable with right click on their label 
- add a searchbar for finding bookmarks

3Domse3, 2022-02-22:
And more:
- make bookmark colors and/or category colors changeable

3Domse3, 2022-02-22:
- make sliders scrollable with mousewheel

3Domse3, 2022-02-22:
- make baseband recording bandwidth selectable

shuyuan-liu, 2022-02-22:
Thanks @3Domse3 for the suggestions. I'd suggest splitting them into separate Issues, maybe grouping the bookmark-related features together, because this thread can get difficult to manage with so many points. I think it will be easier to track progress and discuss each feature in its own Issue than in a single list.

3Domse3, 2022-02-22:
Will do :)

closed 2022-02-22 by 3Domse3