AlexandreRouma/SDRPlusPlus #664: [Improvement request] Change LO frequency automatically after changing frequency via Rigctl

n2iw, 2022-02-21:
I wrote a [script](https://github.com/n2iw/cat-relay) to sync frequency between SDR++ and logging software  RUMlogNG or DXLab Commander. It works fine except when not in "center mode" (target/bullseye), if I change band, the LO is not changed, resulting current frequency out of SDR's coverage. I have to click "bullseye" then click back (I don't want my waterfall to shift when changing frequency on radio). If LO can be changed automatically whenever frequency is out of SDR's coverage, that will be very helpful!!!

BTW. when directly changing frequency in SDR++, LO is changed appropriately no matter in "center mode" or not.

AlexandreRouma, 2022-02-21:
That is simply expected behavior. RigCTL controls the VFO, NOT the center frequency of the SDR.

n2iw, 2022-02-21:
> That is simply expected behavior. RigCTL controls the VFO, NOT the center frequency of the SDR.

Thanks Alex! I dug a little deeper and found out,  LO is changed actually, but it's a little too far away from VFO frequency sometimes. I noticed that at least for Airspy HF+ (I suspect it's similar for other SDRs), sensitivity drops on the edge of the coverage, makes about 80KHz on each edge not very usable.  And a lot of the time (for me), my requested frequency lands near the edge of the bandwidth of the SDR. 

I don't know current algorithm SDR++ used to decide center frequency, but may I suggest if a new center frequency is needed, make it close to or at requested frequency or middle of requested band?

AlexandreRouma, 2022-02-21:
The algorithm only separates the VFO from the border when in normal tuning mode and if the center frequency changes. It doesn't try to prevent the VFO tro go right at the border since the user might want to do it for once reason or another.
What you want is simply, to leave the software in center tuning mode since that won't be an issue for an SDR like the HF+ which doesn't have a DC spike.

n2iw, 2022-02-21:
Yes, using center tuning mode would solve the issue for sure. Except when I tune inside the band, I don't want my waterfall to shift left and right. When waterfall shifts, I lose track of where the stations were, which it the main reason for using waterfall. 

Alternatively, is there a RitCTL command that I can use to change center frequency? If so, I can do it in my script.

AlexandreRouma, 2022-02-21:
Unfortunately no, the RigCTL protocol doesn't have such command. It was designed for normal rigs that don't have a concept of a separate LO, you just tune wherever you want to tune.

You might want to look into writing a module that uses internal functions to do what you're trying to achieve.

closed 2022-09-13 by AlexandreRouma