AlexandreRouma/SDRPlusPlus #665: [Feature request] Baseband recording and replay features

3Domse3, 2022-02-22:
- Baseband replay controls for scrubbing and pausing and recording info view with length, frequency etc
- make baseband recording bandwidth selectable

shuyuan-liu, 2022-02-22:
> make baseband recording bandwidth selectable

Currently this is possible through 1) decimation in the Source tab, or 2) recording "audio" with the RAW demodulator (giving a sample rate equal to your audio sample rate, 44.1kHz or 48kHz usually). The second method requires you to add another Radio module to be able to hear demodulated audio when recording.

> Baseband replay controls for scrubbing and pausing and recording info view with length, frequency etc

I'll add that an "overview waterfall" display of the whole recording like the one in SDR#'s File Player plugin would be very helpful for quickly jumping to the start and end of transmissions. This feature, and seeking, were suggested in #129.

3Domse3, 2022-02-22:
Thank for the reply :) (And sorry for the spam)

AlexandreRouma, 2022-10-04:
combining with #129 since 2 has always been available. Follow that issue instead

closed 2022-10-04 by AlexandreRouma