AlexandreRouma/SDRPlusPlus #666: [Feature request] Bookmark features

3Domse3, 2022-02-22:
- make bookmarks editable with right click on their label
- add a searchbar for finding bookmarks
- make bookmark colors and/or category colors changeable
- bookmarks shouldn't be ordered case-sensitive and numbers should be sorted by their values, not digits (in my opinion)

TarLink48, 2022-02-22:
I support it !

AlexandreRouma, 2022-10-10:
* editable bookmarks, maybe with double click or CTRL+Click but it absolutely should not get in the way tuning VFOs
* searchbar planned
* colors will not be editable, it's an issue with themes.
* bookmark order is set by the C++ sort function and numbers are never sorted so I'm not sure what you're talking about.