AlexandreRouma/SDRPlusPlus #669: changing antenna issue with sdrplay

bobasaurus, 2022-02-23:
When using the SDRplay RSP2 I noticed a problem in SDR++ with the antenna selection.  It defaults to antenna Port A which works fine, changing to antenna Port B in the UI works fine, but changing back to antenna Port A doesn't do anything... it appears to just stay on B.  

I'm using the latest release as of this writing, v1.0.4, on windows 10 x64.

AlexandreRouma, 2022-02-23:
I see other users can replicate. Don't have access to a RSP2 atm so will take a while to fix

nybtor, 2022-02-23:
Not able to reproduce this error on RSP2 nor RSPDX.
Tested on v1.0.4 and v1.0.6 built Feb.18.2022
All win10/11 x64

AlexandreRouma, 2022-02-24:
I'm worried it might be related to the new version API, will have to do more testing. Got a remote mchine with a RSP2 so i'll be able to debug that.

AlexandreRouma, 2022-02-24:
Just tried it myself and I'm unable to replicate the issue as well

AlexandreRouma, 2022-09-13:
Closing as either resolved or an occasional bug in the SDRplay API.