AlexandreRouma/SDRPlusPlus #674: Crashed on macos

vvzvlad, 2022-02-28:
When I press the play button, the program closes after a couple of seconds. 

```
[2022-03-01 00:43:54.099] [info] Ready.
Found Rafael Micro R820T tuner
[2022-03-01 00:44:33.254] [info] RTL-SDR Sample Rate: 2160000.0
Exact sample rate is: 2160000.051498 Hz
[R82XX] PLL not locked!
Disabled direct sampling mode
[2022-03-01 00:44:33.912] [info] RTLSDRSourceModule 'RTL-SDR Source': Start!
[2022-03-01 00:44:33.973] [info] RTLSDRSourceModule 'RTL-SDR Source': Tune: 100100000.0!
fish: Job 1, 'sdrpp' terminated by signal SIGSEGV (Address boundary error)
```

Here is the message from the operating system:

https://pastebin.com/p4S2E8qr

gvanem, 2022-03-01:
Looks like you are out of heap-space in the waterfall-update. Turn it off.

AlexandreRouma, 2022-03-01:
You didn't say what version  of SDR++ you're running, but I'll assume it's probably the old .pkg
In which case, don't use it, get the nightly build: https://www.sdrpp.org/nightly

vvzvlad, 2022-03-09:
[2022-03-09 23:10:05.818] [info] SDR++ v1.0.4

AlexandreRouma, 2022-03-10:
old version, use the nightly

bvogt3000, 2022-04-05:
When I stop my Hack RF on MacOS it crashes the program. I am using the nightly build version 1.0.6 for Intel.

closed 2022-09-29 by AlexandreRouma