AlexandreRouma/SDRPlusPlus #675: Unable to use sdrpp

Kllngii, 2022-03-01:
When running the suggested cmake command for building under macOS `cmake .. \    
  -DOPT_BUILD_AUDIO_SINK=OFF \
  -DOPT_BUILD_BLADERF_SOURCE=OFF \
  -DOPT_BUILD_M17_DECODER=ON \
  -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON \
  -DOPT_BUILD_PLUTOSDR_SOURCE=OFF \
  -DOPT_BUILD_PORTAUDIO_SINK=ON \
  -DOPT_BUILD_SOAPY_SOURCE=OFF` I get an error that the required package glfw3 was not found even though it has been installed via brew and `brew list glfw` verifies that. 

Next I tried installing SDR++ using the provided installer, but starting `srdpp` did not work because the library glew was not found even though it also had been installed. I was able to fix this issue by manually copying the libGLEW.2.2.dylib from the /opt/homebrew/Cellar/ folder to /usr/local/lib but unfortunately I just got another error, because I got the arm64 version of the library not the x86 version (I am using a recent MacBook with M1-Pro chip).

Is there any way to run the current version of SDR++ on a M1 device?

kostecky, 2022-03-01:
Hey @Kllngii - I have a built version in the repo I forked that you can access here: https://github.com/kostecky/SDRPlusPlus/tree/master/SDR%2B%2B.app

I build this every time there are new commits to master. Unfortunately, M1 runner targets are not officially supported as Github actions yet, so I have to do this manually at the moment.

If you move it into your /Applications/ folder, it should "just work". YMMV.

dd5fz, 2022-03-01:
Unfortunately I'm getting a series of errors when I try starting SDR++ :

`
toby@Toby-MacBook-M1 SDR % /Applications/SDR/SDR++.app/Contents/MacOS/sdrpp
[2022-03-01 18:56:23.208] [info] SDR++ v1.0.5
[2022-03-01 18:56:23.211] [info] Loading config
[2022-03-01 18:56:23.212] [warning] ConfigManager locked, waiting...
[2022-03-01 18:56:23.463] [info] Loading icons
[2022-03-01 18:56:23.511] [info] Loading band plans
[2022-03-01 18:56:23.517] [info] Loading band plans color table
[2022-03-01 18:56:23.533] [info] Loading modules
[2022-03-01 18:56:23.533] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_sdr_source.dylib
[2022-03-01 18:56:23.551] [error] Couldn't load /usr/local/lib/sdrpp/plugins/rtl_sdr_source.dylib.
[2022-03-01 18:56:23.551] [info] Loading /usr/local/lib/sdrpp/plugins/meteor_demodulator.dylib
[2022-03-01 18:56:23.566] [error] Couldn't load /usr/local/lib/sdrpp/plugins/meteor_demodulator.dylib.
[2022-03-01 18:56:23.566] [info] Loading /usr/local/lib/sdrpp/plugins/airspy_source.dylib
[2022-03-01 18:56:23.584] [error] Couldn't load /usr/local/lib/sdrpp/plugins/airspy_source.dylib.
[2022-03-01 18:56:23.584] [info] Loading /usr/local/lib/sdrpp/plugins/network_sink.dylib
[2022-03-01 18:56:23.600] [error] Couldn't load /usr/local/lib/sdrpp/plugins/network_sink.dylib.
[2022-03-01 18:56:23.600] [info] Loading /usr/local/lib/sdrpp/plugins/radio.dylib
[2022-03-01 18:56:23.626] [error] Couldn't load /usr/local/lib/sdrpp/plugins/radio.dylib.
[2022-03-01 18:56:23.626] [info] Loading /usr/local/lib/sdrpp/plugins/rigctl_server.dylib
[2022-03-01 18:56:23.627] [error] Couldn't load /usr/local/lib/sdrpp/plugins/rigctl_server.dylib.
[2022-03-01 18:56:23.627] [info] Loading /usr/local/lib/sdrpp/plugins/audio_sink.dylib
[2022-03-01 18:56:23.642] [error] Couldn't load /usr/local/lib/sdrpp/plugins/audio_sink.dylib.
[2022-03-01 18:56:23.642] [info] Loading /usr/local/lib/sdrpp/plugins/m17_decoder.dylib
[2022-03-01 18:56:23.659] [error] Couldn't load /usr/local/lib/sdrpp/plugins/m17_decoder.dylib.
[2022-03-01 18:56:23.659] [info] Loading /usr/local/lib/sdrpp/plugins/frequency_manager.dylib
[2022-03-01 18:56:23.679] [error] Couldn't load /usr/local/lib/sdrpp/plugins/frequency_manager.dylib.
[2022-03-01 18:56:23.679] [info] Loading /usr/local/lib/sdrpp/plugins/recorder.dylib
[2022-03-01 18:56:23.693] [error] Couldn't load /usr/local/lib/sdrpp/plugins/recorder.dylib.
[2022-03-01 18:56:23.693] [info] Loading /usr/local/lib/sdrpp/plugins/file_source.dylib
[2022-03-01 18:56:23.717] [error] Couldn't load /usr/local/lib/sdrpp/plugins/file_source.dylib.
[2022-03-01 18:56:23.717] [info] Loading /usr/local/lib/sdrpp/plugins/airspyhf_source.dylib
[2022-03-01 18:56:23.719] [error] Couldn't load /usr/local/lib/sdrpp/plugins/airspyhf_source.dylib.
[2022-03-01 18:56:23.719] [info] Loading /usr/local/lib/sdrpp/plugins/spyserver_source.dylib
[2022-03-01 18:56:23.735] [error] Couldn't load /usr/local/lib/sdrpp/plugins/spyserver_source.dylib.
[2022-03-01 18:56:23.735] [info] Loading /usr/local/lib/sdrpp/plugins/new_portaudio_sink.dylib
[2022-03-01 18:56:23.751] [error] Couldn't load /usr/local/lib/sdrpp/plugins/new_portaudio_sink.dylib.
[2022-03-01 18:56:23.751] [info] Loading /usr/local/lib/sdrpp/plugins/discord_integration.dylib
[2022-03-01 18:56:23.771] [error] Couldn't load /usr/local/lib/sdrpp/plugins/discord_integration.dylib.
[2022-03-01 18:56:23.771] [info] Loading /usr/local/lib/sdrpp/plugins/weather_sat_decoder.dylib
[2022-03-01 18:56:23.788] [error] Couldn't load /usr/local/lib/sdrpp/plugins/weather_sat_decoder.dylib.
[2022-03-01 18:56:23.788] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_tcp_source.dylib
[2022-03-01 18:56:23.801] [error] Couldn't load /usr/local/lib/sdrpp/plugins/rtl_tcp_source.dylib.
[2022-03-01 18:56:23.801] [info] Loading /usr/local/lib/sdrpp/plugins/hackrf_source.dylib
[2022-03-01 18:56:23.819] [error] Couldn't load /usr/local/lib/sdrpp/plugins/hackrf_source.dylib.
[2022-03-01 18:56:23.819] [info] Initializing Airspy Source (airspy_source)
[2022-03-01 18:56:23.836] [error] Module 'airspy_source' doesn't exist
[2022-03-01 18:56:23.836] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-03-01 18:56:23.848] [error] Module 'airspyhf_source' doesn't exist
[2022-03-01 18:56:23.848] [info] Initializing Audio Sink (audio_sink)
[2022-03-01 18:56:23.867] [error] Module 'audio_sink' doesn't exist
[2022-03-01 18:56:23.867] [info] Initializing BladeRF Source (bladerf_source)
[2022-03-01 18:56:23.883] [error] Module 'bladerf_source' doesn't exist
[2022-03-01 18:56:23.883] [info] Initializing File Source (file_source)
[2022-03-01 18:56:23.900] [error] Module 'file_source' doesn't exist
[2022-03-01 18:56:23.900] [info] Initializing Frequency Manager (frequency_manager)
[2022-03-01 18:56:23.915] [error] Module 'frequency_manager' doesn't exist
[2022-03-01 18:56:23.915] [info] Initializing HackRF Source (hackrf_source)
[2022-03-01 18:56:23.933] [error] Module 'hackrf_source' doesn't exist
[2022-03-01 18:56:23.933] [info] Initializing LimeSDR Source (limesdr_source)
[2022-03-01 18:56:23.953] [error] Module 'limesdr_source' doesn't exist
[2022-03-01 18:56:23.953] [info] Initializing Network Sink (network_sink)
[2022-03-01 18:56:23.965] [error] Module 'network_sink' doesn't exist
[2022-03-01 18:56:23.965] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-03-01 18:56:23.984] [error] Module 'plutosdr_source' doesn't exist
[2022-03-01 18:56:23.984] [info] Initializing RFspace Source (rfspace_source)
[... cut ...]
[2022-03-01 18:56:24.324] [info] Loading /usr/local/share/sdrpp/colormaps/classic_green.json
[2022-03-01 18:56:24.329] [info] New DSP samplerate: 8000000.0 (source samplerate is 8000000.0)
[2022-03-01 18:56:24.329] [info] Updating FFT DSP settings: Keep: 65536, Skip: 334464
[2022-03-01 18:56:24.334] [info] Updating FFT DSP settings: Keep: 65536, Skip: 334464
[2022-03-01 18:56:24.334] [info] Updating FFT DSP settings: Keep: 65536, Skip: 334464
[2022-03-01 18:56:24.335] [info] Updating FFT DSP settings: Keep: 65536, Skip: 334464
[2022-03-01 18:56:24.336] [info] Ready.
`

AlexandreRouma, 2022-03-01:
@dd5fz please create your own issue
@Kllngii here is a M1 build: https://drive.google.com/file/d/1qABrYD8eEVSmk6bv5GvnFt4v6oBQmXlh/view?usp=sharing

dd5fz, 2022-03-01:
Thank you. 

I would have opened an issue at github.com/kostecky if possible.

The newest M1 build appears to be working with a RTL-SDR stick on my MacBook Pro (Apple M1 Max, 16-inch) running Monterey 12.2.1.

Kllngii, 2022-03-01:
Thanks @kostecky and @AlexandreRouma!

@kostecky your code executed flawlessly however it did not find my sdr.

@AlexandreRouma your code also worked flawlessly and it did show my device. Now everything is working fine.

Would you mind telling me how exactly you managed building the app for M1 devices successfully?

kostecky, 2022-03-01:
> Thank you.
> 
> I would have opened an issue at github.com/kostecky if possible.
> 
> The newest M1 build appears to be working with a RTL-SDR stick on my MacBook Pro (Apple M1 Max, 16-inch) running Monterey 12.2.1.

Issues are enabled now. I'll fix this up later on. In the meantime I see a working build has been posted.

closed 2022-09-13 by AlexandreRouma