AlexandreRouma/SDRPlusPlus #676: No audio on Ubuntu 20.04

scottbomb, 2022-03-03:
Hi, I just installed it as you have instructed at https://github.com/AlexandreRouma/SDRPlusPlus#debian-based-ubuntu-mint-etc

I can use the tuner and see signals, waterfall works, "Radio" is checked, and yet I'm not getting any sound. Using RTL-SDR, Realtek RTL2838.

shuyuan-liu, 2022-03-03:
Make sure in the Sinks tab you have changed "None" to "Audio" and selected the correct output device.

scottbomb, 2022-03-03:
> eply-To: AlexandreRouma/SDRPlusPlus <
> scottbomb ***@***.***>, Author ***@***.***>Subj
> #676)Date: Wed, 02 Mar 2022 16:33:16 -0800

Thanks for the tip. I saw Audio was already selected but I had to
change the output from my sound card to "default" for it to work.
Thanks again,Scott

closed 2022-03-03 by AlexandreRouma

John-Telek, 2023-03-25:
I'm having the same issue on Debian 11 with the nightly releases. If I compile my own The speaker icon is not locked out, but the nightly install is locked even though i've got radio selected and audio as the sink.

Any ideas pls ?

Cheers,
    John