AlexandreRouma/SDRPlusPlus #677: Crash / Unclean shutdown on Mac OS (M10 - libc++abi: terminating

jordancrombie, 2022-03-04:
Connecting to a remote SDR++ server instance with an already connected client, the program dumps to shell with the following:

...... Regular Startup ......
[2022-03-03 22:33:50.415] [info] Ready.
[2022-03-03 22:34:27.483] [info] AirspySourceModule 'Airspy Source': Menu Deselect!
[2022-03-03 22:34:27.485] [info] SDRPPServerSourceModule 'SDR++ Server Source': Menu Select!
[2022-03-03 22:34:31.136] [error] Asked to disconnect by the server
libc++abi: terminating
zsh: abort      Downloads/SDR++.app/Contents/MacOS/sdrpp

Not the end of the world, but just logging for tracking....

AlexandreRouma, 2022-03-04:
known issue

drewzh, 2022-09-06:
Also experiencing this issue quite routinely when switching between interfaces on the client (which happens often here, unfortunately).  Same log as above. Any way to track this fix or any hints as to why/where it's occurring so a patch could be independently created? Many thanks

AlexandreRouma, 2024-04-08:
This should now be fixed, let me know if not.

closed 2024-04-08 by AlexandreRouma