AlexandreRouma/SDRPlusPlus #680: Seg faults with new USB library (1.0.25) on Arch

horga83, 2022-03-06:
I have noticed that SDRPP now seg faults on Arch linux with libusb-1.0.25 installed.
If I down grade with:
sudo pacman -U libusb-1.0.24-2-x86_64.pkg.tar.zst
sudo pacman -U lib32-libusb-1.0.24-1-x86_64.pkg.tar.zst

it runs fine.

starvald, 2022-03-06:
Arch user here also, did it crash with an airspy hf?
if so you can try rebuilding airspyhf-git from aur, it was patched to work with libusb 1.0.25
https://github.com/airspy/airspyhf/commit/646141f59101b0fe0fc0b624f8eee13e2de42b44

horga83, 2022-03-06:
No, I'm using a LimeSDR. 
Gqrx works fine with the newer libusb and it uses the same libs to talk
to the Limesdr so not sure where the problem is.

horga83, 2022-03-06:
Here is an strace dump of the last few lines until it seg faults if that helps.

openat(AT_FDCWD, "/sys/bus/usb/devices/4-4.1.4/bConfigurationValue", O_RDONLY|O_CLOEXEC) = 14
read(14, "1\n", 19)                     = 2
close(14)                               = 0
openat(AT_FDCWD, "/sys/bus/usb/devices/4-4.1.4/bConfigurationValue", O_RDONLY|O_CLOEXEC) = 14
read(14, "1\n", 19)                     = 2
close(14)                               = 0
timerfd_settime(8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=424702, tv_nsec=775397701}}, NULL) = 0
ioctl(13, USBDEVFS_SUBMITURB, 0x564b97892230) = 0
futex(0x564b97871d90, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=424762, tv_nsec=675437850}, FUTEX_BITSET_MATCH_ANY) = 0
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 0
write(2, "TransferPacket: Write failed (re"..., 38TransferPacket: Write failed (ret=-1)
) = 38
write(2, "Failed to reset USB bulk endpoin"..., 35Failed to reset USB bulk endpoints
) = 35
timerfd_settime(8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=424702, tv_nsec=779023048}}, NULL) = 0
ioctl(13, USBDEVFS_SUBMITURB, 0x564b97892230) = 0
futex(0x564b97871d94, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=424762, tv_nsec=679071553}, FUTEX_BITSET_MATCH_ANY) = 0
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 0
write(2, "TransferPacket: Write failed (re"..., 38TransferPacket: Write failed (ret=-1)
) = 38
timerfd_settime(8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=424702, tv_nsec=782741392}}, NULL) = 0
ioctl(13, USBDEVFS_SUBMITURB, 0x564b97892230) = 0
futex(0x564b97871d90, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=424762, tv_nsec=682782974}, FUTEX_BITSET_MATCH_ANY) = 0
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 0
write(2, "TransferPacket: Write failed (re"..., 38TransferPacket: Write failed (ret=-1)
) = 38
timerfd_settime(8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=424702, tv_nsec=786454657}}, NULL) = 0
ioctl(13, USBDEVFS_SUBMITURB, 0x564b97892230) = 0
futex(0x564b97871d94, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=424762, tv_nsec=686499122}, FUTEX_BITSET_MATCH_ANY) = 0
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 0
write(2, "TransferPacket: Write failed (re"..., 38TransferPacket: Write failed (ret=-1)
) = 38
timerfd_settime(8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=424702, tv_nsec=789969325}}, NULL) = 0
ioctl(13, USBDEVFS_SUBMITURB, 0x564b97892230) = 0
futex(0x564b97871d90, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=424762, tv_nsec=690015458}, FUTEX_BITSET_MATCH_ANY) = 0
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 0
write(2, "TransferPacket: Write failed (re"..., 37TransferPacket: Write failed (ret=0)
) = 37
timerfd_settime(8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=424702, tv_nsec=794021819}}, NULL) = 0
ioctl(13, USBDEVFS_SUBMITURB, 0x564b978b10a0) = 0
futex(0x564b97871d94, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=424762, tv_nsec=694104515}, FUTEX_BITSET_MATCH_ANY) = 0
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 0
write(2, "TransferPacket: Write failed (re"..., 37TransferPacket: Write failed (ret=0)
) = 37
timerfd_settime(8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=424702, tv_nsec=797815197}}, NULL) = 0
ioctl(13, USBDEVFS_SUBMITURB, 0x564b978b10a0) = 0
futex(0x564b97871d90, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=424762, tv_nsec=697949583}, FUTEX_BITSET_MATCH_ANY) = 0
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 0
write(2, "TransferPacket: Write failed (re"..., 37TransferPacket: Write failed (ret=0)
) = 37
timerfd_settime(8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=424702, tv_nsec=801527431}}, NULL) = 0
ioctl(13, USBDEVFS_SUBMITURB, 0x564b978b10a0) = 0
futex(0x564b97871d94, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=424762, tv_nsec=701577684}, FUTEX_BITSET_MATCH_ANY) = 0
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 0
write(2, "TransferPacket: Write failed (re"..., 38TransferPacket: Write failed (ret=-1)
) = 38
timerfd_settime(8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=424702, tv_nsec=805216075}}, NULL) = 0
ioctl(13, USBDEVFS_SUBMITURB, 0x564b978b10a0) = 0
futex(0x564b97871d90, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=424762, tv_nsec=705257851}, FUTEX_BITSET_MATCH_ANY) = 0
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 0
write(2, "TransferPacket: Write failed (re"..., 37TransferPacket: Write failed (ret=0)
) = 37
ioctl(13, USBDEVFS_RELEASEINTERFACE, 0x7ffd68beaf34) = 0
write(7, "\1\0\0\0\0\0\0\0", 8)         = 8
futex(0x564b97871d98, FUTEX_WAKE_PRIVATE, 1) = 1
close(13)                               = 0
futex(0x564b97871d94, FUTEX_WAKE_PRIVATE, 2147483647) = 1
futex(0x564b97871d40, FUTEX_WAKE_PRIVATE, 1) = 1
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x40} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

AlexandreRouma, 2022-03-06:
Not a SDR++ issue, LibUSB 1.0.25 exposed some bugs in the SDR libraries.
Rollback to libusb 1.0.24 and refer to the SDR driver authors for future fixes.

closed 2022-03-06 by AlexandreRouma

horga83, 2022-03-07:
Ok thank you.