AlexandreRouma/SDRPlusPlus #682: After Rebuild of librtaudio-dev, SDR++ is crashing

gbeyer3, 2022-03-09:
Hi, I followed the steps in issue #112 to build a new librtaudio-dev.  Using 5.1.0 branch all seemed to work. But when I invoked SDR++ it crashed.  Invoking it from CLI, I find the following errors (see my <<<<<<<<<  call-outs:

BTW - rtl_test is coming back OK.

$ ./sdrpp
[2022-03-09 15:51:27.225] [info] SDR++ v1.0.4
[2022-03-09 15:51:27.226] [info] Loading config
[2022-03-09 15:51:27.313] [info] Using OpenGL 3.0
[2022-03-09 15:51:27.383] [info] Loading icons
[2022-03-09 15:51:27.388] [info] Loading band plans
[2022-03-09 15:51:27.389] [info] Loading band plans color table
[2022-03-09 15:51:27.392] [info] Loading modules
[2022-03-09 15:51:27.392] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2022-03-09 15:51:27.392] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2022-03-09 15:51:27.393] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2022-03-09 15:51:27.394] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-03-09 15:51:27.411] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-03-09 15:51:27.429] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2022-03-09 15:51:27.447] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2022-03-09 15:51:27.462] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2022-03-09 15:51:27.479] [info] Loading /usr/lib/sdrpp/plugins/weather_sat_decoder.so
[2022-03-09 15:51:27.496] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2022-03-09 15:51:27.512] [info] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[2022-03-09 15:51:27.530] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2022-03-09 15:51:27.546] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2022-03-09 15:51:27.562] [error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.    <<<<<<<<
[2022-03-09 15:51:27.562] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2022-03-09 15:51:27.579] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-03-09 15:51:27.596] [info] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[2022-03-09 15:51:27.614] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2022-03-09 15:51:27.629] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2022-03-09 15:51:27.646] [info] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[2022-03-09 15:51:27.662] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2022-03-09 15:51:27.679] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2022-03-09 15:51:27.696] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2022-03-09 15:51:27.712] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2022-03-09 15:51:27.729] [info] Initializing Airspy Source (airspy_source)
[2022-03-09 15:51:27.749] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-03-09 15:51:27.765] [info] Initializing Audio Sink (audio_sink)
[2022-03-09 15:51:27.778] [info] Initializing BladeRF Source (bladerf_source)
[2022-03-09 15:51:27.799] [error] Could not list devices -7                                              <<<<<<<<<<
[2022-03-09 15:51:27.799] [info] Initializing File Source (file_source)
[2022-03-09 15:51:27.811] [info] Initializing Frequency Manager (frequency_manager)
[2022-03-09 15:51:27.828] [info] Initializing HackRF Source (hackrf_source)
[2022-03-09 15:51:27.849] [info] Initializing LimeSDR Source (limesdr_source)
[2022-03-09 15:51:27.868] [info] Initializing Network Sink (network_sink)
[2022-03-09 15:51:27.878] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-03-09 15:51:27.895] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Found Rafael Micro R820T tuner
[2022-03-09 15:51:28.376] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-03-09 15:51:28.377] [info] Initializing Radio (radio)

gbeyer3, 2022-03-09:
#507

AlexandreRouma, 2022-03-10:
RtAudio issue, not SDR++
You probably compiled it wrong.

closed 2022-03-10 by AlexandreRouma