AlexandreRouma/SDRPlusPlus #684: crashing on windows 10 when starting up

davidsaroff, 2022-03-11:
I'm trying to download and run SDR++ on a windows laptop. I find the most recent version on github at URL

https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/1.0.4/sdrpp_windows_x64.zip

I download the zip, unzip it to a folder on my desktop and double click on sdrpp.exe 

A window appears, the program appears to be initializing, That initialization window closes. Nothing more happens. If I open the Task Manager I can see sdrpp.exe appear, run for a second, then disappear from the Task Manager list.

The windows is automatically updated. The current version is21H2 (OS Build 19044.1586)
 
Any help would be appreciated.
David SaroffKM4DCQ

shuyuan-liu, 2022-03-11:
Could you run sdrpp.exe from a command prompt with the `-s` option and post the debug messages produced?

davidsaroff, 2022-03-11:
Alexandre,

This is what I see in a windows power shell

PS C:\Users\david>
C:\Users\david\Desktop\sdrpp_windows_x64\sdrpp_windows_x64\sdrpp.exe
[2022-03-11 17:01:08.542] [info] SDR++ v1.0.4

PS C:\Users\david>
C:\Users\david\Desktop\sdrpp_windows_x64\sdrpp_windows_x64\sdrpp.exe -s
[2022-03-11 17:01:17.981] [info] SDR++ v1.0.4
[2022-03-11 17:01:17.982] [info] Loading config
[2022-03-11 17:01:17.983] [error] Config file './config.json' is corrupted,
resetting it
[2022-03-11 17:01:17.987] [error] Resource directory doesn't exist! Please
make sure that you've configured it correctly in config.json (check readme
for details)

David

AlexandreRouma, 2022-03-11:
Sounds like you're running it from somewhere where it doesn't have write permission

davidsaroff, 2022-03-11:
Alexandre,

I showed you a try from a windows command prompt.
Opening a WSL ubuntu 18.04 window I try running it.
This is what I see

***@***.***:/mnt/c/Users/david/Desktop/sdrpp_windows_x64/sdrpp_windows_x64$
ll sdrpp.exe
-rwxrwxrwx 1 dsaroff dsaroff 45056 Mar 11 17:12 sdrpp.exe*

***@***.***:/mnt/c/Users/david/Desktop/sdrpp_windows_x64/sdrpp_windows_x64$
./sdrpp.exe -s
[2022-03-11 17:25:51.850] [info] SDR++ v1.0.4
[2022-03-11 17:25:51.853] [info] Loading config
[2022-03-11 17:25:52.173] [info] Using OpenGL 3.0
[2022-03-11 17:25:52.334] [info] Loading icons
[2022-03-11 17:25:52.358] [info] Loading band plans
[2022-03-11 17:25:52.363] [info] Loading band plans color table
[2022-03-11 17:25:52.368] [info] Loading modules
[2022-03-11 17:25:52.369] [info] Loading ./modules/airspyhf_source.dll
[2022-03-11 17:25:52.380] [info] Loading ./modules/airspy_source.dll
[2022-03-11 17:25:52.384] [info] Loading ./modules/audio_sink.dll
[2022-03-11 17:25:52.394] [info] Loading ./modules/bladerf_source.dll
[2022-03-11 17:25:52.398] [info] Loading ./modules/discord_integration.dll
[2022-03-11 17:25:52.401] [info] Loading ./modules/file_source.dll
[2022-03-11 17:25:52.405] [info] Loading ./modules/frequency_manager.dll
[2022-03-11 17:25:52.409] [info] Loading ./modules/hackrf_source.dll
[2022-03-11 17:25:52.416] [info] Loading ./modules/limesdr_source.dll
[2022-03-11 17:25:52.423] [info] Loading ./modules/m17_decoder.dll
[2022-03-11 17:25:52.432] [info] Loading ./modules/meteor_demodulator.dll
[2022-03-11 17:25:52.436] [info] Loading ./modules/network_sink.dll
[2022-03-11 17:25:52.440] [info] Loading ./modules/plutosdr_source.dll
[2022-03-11 17:25:52.447] [info] Loading ./modules/radio.dll
[2022-03-11 17:25:52.451] [info] Loading ./modules/recorder.dll
[2022-03-11 17:25:52.457] [info] Loading ./modules/rigctl_server.dll
[2022-03-11 17:25:52.463] [info] Loading ./modules/rtl_sdr_source.dll
[2022-03-11 17:25:52.469] [info] Loading ./modules/rtl_tcp_source.dll
[2022-03-11 17:25:52.474] [info] Loading ./modules/sdrplay_source.dll
[2022-03-11 17:25:52.481] [info] Loading ./modules/soapy_source.dll
[2022-03-11 17:25:52.485] [info] Loading ./modules/spyserver_source.dll
[2022-03-11 17:25:52.490] [info] Initializing Airspy Source (airspy_source)
[2022-03-11 17:25:52.545] [info] Initializing AirspyHF+ Source
(airspyhf_source)
[2022-03-11 17:25:52.593] [info] Initializing Audio Sink (audio_sink)
[2022-03-11 17:25:52.594] [info] Initializing BladeRF Source
(bladerf_source)
[2022-03-11 17:25:52.642] [error] Could not list devices -7
[2022-03-11 17:25:52.643] [info] Initializing File Source (file_source)
[2022-03-11 17:25:52.644] [info] Initializing Frequency Manager
(frequency_manager)
[2022-03-11 17:25:52.645] [info] Initializing HackRF Source (hackrf_source)
[2022-03-11 17:25:52.700] [info] Initializing LimeSDR Source
(limesdr_source)
[2022-03-11 17:25:52.703] [info] Initializing Network Sink (network_sink)
[2022-03-11 17:25:52.705] [info] Initializing PlutoSDR Source
(plutosdr_source)
[2022-03-11 17:25:52.708] [info] Initializing RTL-SDR Source
(rtl_sdr_source)
[2022-03-11 17:25:52.754] [info] Initializing RTL-TCP Source
(rtl_tcp_source)
[2022-03-11 17:25:52.757] [info] Initializing Radio (radio)
[2022-03-11 17:25:52.850] [info] RtAudio stream open
[2022-03-11 17:25:52.851] [info] Initializing Recorder (recorder)
[2022-03-11 17:25:52.853] [info] Initializing Rigctl Server (rigctl_server)
[2022-03-11 17:25:52.855] [info] Initializing SDRplay Source
(sdrplay_source)
[2022-03-11 17:25:52.858] [info] Initializing SoapySDR Source (soapy_source)
[ERROR] SoapySDR::loadModule(C:\Program
Files\PothosSDR/lib/SoapySDR/modules0.6/LMS7Support.dll)
  LoadLibrary() failed: The specified procedure could not be found.

Win32; Microsoft Visual C++ version 14.0; Boost_106300;
UHD_003.010.002.000-3-g122bfae1

***@***.***
:/mnt/c/Users/david/Desktop/sdrpp_windows_x64/sdrpp_windows_x64$

Thanks for looking at this.

David Saroff

AlexandreRouma, 2022-03-11:
you cannot run SDR++ in wsl

davidsaroff, 2022-03-11:
in command prompt

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Microsoft Windows [Version 10.0.19044.1586]
(c) Microsoft Corporation. All rights reserved.

C:\Users\david>cd Desktop\sdrpp_windows_x64\sdrpp_windows_x64\

C:\Users\david\Desktop\sdrpp_windows_x64\sdrpp_windows_x64>dir
 Volume in drive C is OS
 Volume Serial Number is 9863-741C

 Directory of C:\Users\david\Desktop\sdrpp_windows_x64\sdrpp_windows_x64

03/11/2022  05:14 PM    <DIR>          .
03/11/2022  05:14 PM    <DIR>          ..
03/11/2022  05:12 PM            59,392 airspy.dll
03/11/2022  05:12 PM           110,592 airspyhf.dll
03/11/2022  05:14 PM                42 airspyhf_config.json
03/11/2022  05:14 PM                42 airspy_config.json
03/11/2022  05:14 PM                 2 audio_sink_config.json
03/11/2022  05:12 PM           720,384 bladeRF.dll
03/11/2022  05:14 PM                42 bladerf_config.json
03/11/2022  05:25 PM             3,853 config.json
03/11/2022  05:12 PM           927,232 fftw3f.dll
03/11/2022  05:14 PM                20 file_source_config.json
03/11/2022  05:25 PM               190 frequency_manager_config.json
03/11/2022  05:12 PM           503,808 glew32.dll
03/11/2022  05:12 PM           226,304 glfw3.dll
03/11/2022  05:12 PM            54,272 hackrf.dll
03/11/2022  05:14 PM                42 hackrf_config.json
03/11/2022  05:12 PM           368,128 libad9361.dll
03/11/2022  05:12 PM         1,872,896 libcodec2.dll
03/11/2022  05:12 PM         1,653,760 libiio.dll
03/11/2022  05:12 PM           129,024 libusb-1.0.dll
03/11/2022  05:14 PM                42 limesdr_config.json
03/11/2022  05:12 PM         1,029,632 LimeSuite.dll
03/11/2022  05:14 PM                 2 m17_decoder_config.json
03/11/2022  05:14 PM                 2 meteor_demodulator_config.json
03/11/2022  05:12 PM    <DIR>          modules
03/11/2022  05:14 PM                 2 network_sink_config.json
03/11/2022  05:14 PM                97 plutosdr_source_config.json
03/11/2022  05:12 PM            78,848 pthreadVC2.dll
03/11/2022  05:14 PM                 2 radio_config.json
03/11/2022  05:14 PM                 2 recorder_config.json
03/11/2022  05:14 PM    <DIR>          recordings
03/11/2022  05:12 PM    <DIR>          res
03/11/2022  05:14 PM                 2 rigctl_server_config.json
03/11/2022  05:12 PM            70,144 rtaudio.dll
03/11/2022  05:12 PM            87,552 rtlsdr.dll
03/11/2022  05:14 PM                41 rtl_sdr_config.json
03/11/2022  05:25 PM               241 rtl_tcp_config.json
03/11/2022  05:12 PM            52,200 sdrplay_api.dll
03/11/2022  05:14 PM                42 sdrplay_config.json
03/11/2022  05:12 PM            45,056 sdrpp.exe
03/11/2022  05:12 PM         2,099,200 sdrpp_core.dll
03/11/2022  05:14 PM                42 soapy_source_config.json
03/11/2022  05:14 PM                72 spyserver_config.json
03/11/2022  05:12 PM         2,660,352 volk.dll
              40 File(s)     12,753,598 bytes
               5 Dir(s)  55,575,273,472 bytes free

C:\Users\david\Desktop\sdrpp_windows_x64\sdrpp_windows_x64>sdrpp.exe -s
[2022-03-11 17:47:36.985] [info] SDR++ v1.0.4
[2022-03-11 17:47:36.987] [info] Loading config
[2022-03-11 17:47:37.326] [info] Using OpenGL 3.0
[2022-03-11 17:47:37.481] [info] Loading icons
[2022-03-11 17:47:37.508] [info] Loading band plans
[2022-03-11 17:47:37.513] [info] Loading band plans color table
[2022-03-11 17:47:37.520] [info] Loading modules
[... 37 lines cut ...]
[2022-03-11 17:47:37.859] [info] Initializing RTL-SDR Source
(rtl_sdr_source)
[2022-03-11 17:47:37.897] [info] Initializing RTL-TCP Source
(rtl_tcp_source)
[2022-03-11 17:47:37.899] [info] Initializing Radio (radio)
[2022-03-11 17:47:37.973] [info] RtAudio stream open
[2022-03-11 17:47:37.974] [info] Initializing Recorder (recorder)
[2022-03-11 17:47:37.977] [info] Initializing Rigctl Server (rigctl_server)
[2022-03-11 17:47:37.978] [info] Initializing SDRplay Source
(sdrplay_source)
[2022-03-11 17:47:37.980] [info] Initializing SoapySDR Source (soapy_source)
←[1m←[31m[ERROR] SoapySDR::loadModule(C:\Program
Files\PothosSDR/lib/SoapySDR/modules0.6/LMS7Support.dll)
  LoadLibrary() failed: The specified procedure could not be found.
←[0m
Win32; Microsoft Visual C++ version 14.0; Boost_106300;
UHD_003.010.002.000-3-g122bfae1

C:\Users\david\Desktop\sdrpp_windows_x64\sdrpp_windows_x64>
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

It looks to me that there is a SoapySDR library from the installation of
another SDR tool that is confusing things.

David Saroff

davidsaroff, 2022-03-11:
Alexandre,

I checked and uninstalled unnecessary and previous tries with SDR tools.

What are installed are
for pluto
IIO Oscilloscope

for SDRplay RSP1
SDRplay SDR Console
SDRplay SDRuno

It still fails the same way, on the SoapySDR library load
[2022-03-11 17:47:37.980] [info] Initializing SoapySDR Source (soapy_source)
←[1m←[31m[ERROR] SoapySDR::loadModule(C:\Program
Files\PothosSDR/lib/SoapySDR/modules0.6/LMS7Support.dll)
  LoadLibrary() failed: The specified procedure could not be found.
←[0m

Your thoughts?

David Saroff
KM4DCQ

AlexandreRouma, 2022-03-11:
your version of SoapySDR is outdated. Install a new version or disable the soapy source module (very likely you won't need it) by moving the soapy_source.dll file out of the modules directory

davidsaroff, 2022-03-12:
Alexandre,

Following your suggestion to update SoapySDR this is the installer that I
tried.

https://downloads.myriadrf.org/builds/PothosSDR/PothosSDR-2021.07.25-vc16-x64.exe

It loads and runs!

Thanks!

David Saroff
KM4DCQ

closed 2022-10-10 by AlexandreRouma

K5WH-Walter, 2023-12-13:
I tried updating the soapy dll, and I no longer het the failure message about soapy, but the application flashes the screen as it's loading all the other dll files, but then disappears.
When I run sdrpp -s everything looks fine with no errors at all.
Any possible solutions to get past this?
Using a RTL-SDR V3 dongle on Win10. 
Thanks, Walter/K5WH