AlexandreRouma/SDRPlusPlus #690: [Feature request] Recorder module - create a new file one every transmission heard on the channel

atlex, 2022-03-15:
Hello @AlexandreRouma !
Thank you for the great tool!

Could you please update the Recorder module to create a new file one every transmission heard on the channel.
It is very useful to have recorded files without any silence. 
An output file can be completed/closed when squelch is closed.

You can se it in action in RTLSDR-Airband  - split_on_transmission property
https://github.com/szpajder/RTLSDR-Airband/wiki/Configuring-file-outputs

There is very good example -  the SDR# Audio Recorder plugin
http://rtl-sdr.ru/page/modificirovannyj-plagin-zapisi-iz-komplekta-sdrsharp

Ta-2om, 2022-03-16:
Actually you can do it with rtl-sdr-scanner (https://github.com/shajen/rtl-sdr-scanner) on Linux. Windows I don't know.
Or we will able to do with SDRpp..But all these feature request will make SDRpp one day no more "bloatfree" but "bloatfull" ;-)

atlex, 2022-04-06:
I think it's very useful to have ability to record some signals and have separate file for every transmission.

AlexandreRouma, 2022-09-15:
There now is an option to not record silence in the latest nightly.
I will not however add an option to create separate files each time silence is detected. You would very quickly end up with thousands of files, it's just not a good idea.

closed 2022-09-15 by AlexandreRouma