AlexandreRouma/SDRPlusPlus #695: Nightly macOS build won't start

petermllrr, 2022-03-19:
I downloaded the nightly build from https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/1872297127 but the app just terminates when launched.

macOS 12.3 (Intel)

Any pointers?

ja-Bertolin, 2022-03-19:
It happens the same for me, if you try three-four times it starts fine, and works fine. I opened an Issue related to that #678

AlexandreRouma, 2022-03-19:
could you try this exact build?
https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/1932808676

ja-Bertolin, 2022-03-19:
This is the one that I am using and opened issue #678 ,in other post you mentioned something about a library that was provoking the crash in macos

pumatrax, 2022-04-15:
Got it installed and running on my macOS. 

Installed these first: brew install fftw glew glfw volk airspy airspyhf portaudio hackrf rtl-sdr libbladerf

received an error about permission denied and then ran this.

sudo chown -R $(whoami) $(brew --prefix)/*

AlexandreRouma, 2022-10-04:
closing as stalled

closed 2022-10-04 by AlexandreRouma