AlexandreRouma/SDRPlusPlus #697: Dependencies to the -dev packages since 1.0.4

flux242, 2022-03-20:
Why do you want -dev packages to be installed? We do not install no -dev packages unless we are developing on them

closed 2022-08-17 by aosync

AlexandreRouma, 2022-08-17:
and I'll fix the deb to not include `-dev` where possible

flux242, 2022-08-20:
where was it completed? Here's the nightly deb package:

Package: sdrpp
Version: 1.1.0-769
Maintainer: Ryzerth
Architecture: all
Description: Bloat-free SDR receiver software
Depends: libfftw3-dev, libglfw3-dev, libvolk2-dev, librtaudio-dev, libzstd-dev

AlexandreRouma, 2022-08-21:
It was closed by mistake by @aosync and as I said I'll fix it later. I don't currently have the time to mess with package names.
it's not as simple as removing the dev suffix.

flux242, 2022-08-22:
What are you talking about? You just provide proper packages names to that darn sh script that generates the deb. Is that the complexity you're talking about?