AlexandreRouma/SDRPlusPlus #698: AirSpy Hf+ stops audio every 14 secods

ja-Bertolin, 2022-03-21:
Hi

Recently, I am experiencing a strange behaviour with my AirSpy Hf+ (after two years of using it) (attached a video that you can see in second 1)

After a long period of using it (almost one day) it starts to stop audio every 14 seconds (not spectrum but only audio) for a second. I have tested GQRX and it seems that it is not present.

Is it normal? It is the first time that I face it.

Thanks in beforehand for your comments

ja-Bertolin, 2022-03-21:
Tested with versions #642 to #619 with the same results, but GQRX does not present it. What could be the reason?

ja-Bertolin, 2022-03-21:
I have rebooted the laptop (MacOS 11.6.5) and now it works fine.

Usually the app for MacOS crashes two-three times before it runs correctly ( I opened issues #647 #681 ) , is it possible that any component of the software remains running after it crashing?

AlexandreRouma, 2023-10-30:
Since no other report has shown up I'll assume this was some power management issue on the mac shutting off the airspy.

closed 2023-10-30 by AlexandreRouma