AlexandreRouma/SDRPlusPlus #699: SDR++ audo problem

delboy85341, 2022-03-22:
I have a Dell Optiplex 7-1- with 32GB RAM.  Kubuntu 20.04 is the OS.  SDR ++  1.0.4 is installed.  Sometimes when I start it, the audio sounds like it's being turned on and off rapidly.  Other times when I start it, the audio is playing like it should.  The audio problem happens when I do something.  For example, if I change the audio sample rate, or if I change the sample rate, say from 10 MHz to 5MHz.  I've found the audio returns to normal if I change the audio sample rate or the device the audio is sent to.  I've found that when I use SDR++ to record the distorted audio, it isn't recorded with the distortion.  If I use audacity to record the distorted audio, the audio in the recording is distorted.  I have a few seconds of audio recorded with SDR++ and audacity at ladmo.asuscomm.com/sdrpp

Is there a way to fix this audio problem?  Thanks.

AlexandreRouma, 2022-03-22:
RtAudio bug:
https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

delboy85341, 2022-03-23:
Thanks for your help.  It works perfectly now.

closed 2022-03-23 by AlexandreRouma