AlexandreRouma/SDRPlusPlus #701: Chinese characters are not supported and displayed“?”

long996, 2022-03-24:
Chinese characters are not supported and displayed“?”

shuyuan-liu, 2022-03-24:
Support for CJK (and other Unicode characters) is not planned. See #275.

AlexandreRouma, 2022-03-24:
after some imgui updates it might work if the font contained chinese characters, but unfortunately, it doesn't.

shuyuan-liu, 2022-03-27:
@AlexandreRouma would an option to select custom fonts (runtime or buildtime) be possible?

AlexandreRouma, 2022-03-27:
Not possible, ImGui relies on font size for scaling and the rest of the code doesn't support that

shuyuan-liu, 2022-03-27:
> Not possible

Tried replacing Roboto-Medium.ttf in the resource directory with Chinese fonts (Microsoft Sim Sun, WQY Micro Hei). The font does get used and nothing looks too broken; see screenshots attached. Would you mind testing / elaborating on why you think it's not possible? 

Sadly, with the new fonts, CJK characters are still `?`. The same `frequency_manager_config.json` as in #275 was used.

[image]

[image]

AlexandreRouma, 2022-03-27:
changing the font isn't the only thing you have to do to make it work. You also have to tell ImGui to load the chinese characters.
Switch to the `backend_abstraction` branchy and add `builder.AddRanges(fonts->GetGlyphRangesChineseFull());` after line 30

bubnikv, 2022-04-06:
This is what we have done in PrusaSlicer to support wide range of languages:

This function initializes the glyph ranges.
https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/ImGuiWrapper.cpp#L130

Also based on the language we need to load the proper font. We currently distribute PrusaSlicer with two fonts for ImGUI rendering: NotoSans-Regular.ttf and NotoSansCJK-Regular.ttc, the latter used for Chinese, Japanese and Korean. The CKJ font is quite big.

https://github.com/prusa3d/PrusaSlicer/blob/6b6fb9f692113b56098d2076339e2d4e136c40fd/src/slic3r/GUI/ImGuiWrapper.cpp#L1239

AlexandreRouma, 2022-09-30:
Support for CJK will not be added is it results in huge ImGui font atlas textures which causes issues on some devices.
I guess pinyin can be used instead?

closed 2022-09-30 by AlexandreRouma

shuyuan-liu, 2022-09-30:
> I guess pinyin can be used instead?

Yes but it would be really awkward. Many characters share the same pinyin, and having no easy way of typing tones only confuses things further as without tones even more characters will have the same pinyin (the classic 妈mā 麻má 马mǎ 骂mà for example). It feels a bit like DOS limiting all filenames to 8+3 chars: it's not impossible to use, but certainly not comfortable either.

I do understand CJK support isn't a priority, but just thought I'd give some perspective on why it would be very useful to have.

AlexandreRouma, 2022-09-30:
the issue is I tried adding support for it, but the resulting texture atlas makes it either crash or use a huge amount of GPU memory. The number of characters is so much higher than latin/cyrillic alphabets. It's a design flaw with ImGui that just can't be dealt with unless they rewrite their text stuff to dynamically load characters instead of needing a full precompiled texture of them.

shuyuan-liu, 2022-09-30:
Thanks for the explanation Alexandre. In that case we'll need to wait for ImGUI to make those changes.

wc7086, 2023-06-16:
Due to a design flaw in imgui, support for cjk characters must await the merging of the following pr

https://github.com/ocornut/imgui/pull/3471