AlexandreRouma/SDRPlusPlus #712: FFT Wraparound

nerdspice, 2022-04-03:
Hey guys, I'm having a strange issue with strong signals wrapping around the edges of the FFT display. SDR++ 1.0.4 - Win10 using RTL-SDR v3. This issue doesn't appear in SDR#

https://user-images.githubusercontent.com/49222223/161418906-27d9356b-a753-4b00-92c6-1ee1ac50394b.mp4

AlexandreRouma, 2022-04-04:
Gonna bet on a bad source config option like way too high gain but since you're not showing the source menu I can't really help...

nerdspice, 2022-04-04:
Sorry, I should've shown the source, you're right. I'll take another video. I use the same tuner options in SDR#, if that helps

https://user-images.githubusercontent.com/49222223/161609543-3570a865-23d8-4d65-9ac2-72e2f151db4b.mp4

.

nerdspice, 2022-04-04:
Lemme know if there's other config options I could try.

AlexandreRouma, 2022-04-04:
The settings look correct
So, the thing is, if you check what you see in SDR#, you'll notice that quite a bit of the sides are cut off (set it to he same samplerate as SDR++, set the zoom in SDR# to the minimum at looks where the sides end up). SDR++ doesn't hide the side where some images can occure, so that's why you're seeing it.
It can also happen when the gain is too higg but from the source settings it doesn't look like it

nerdspice, 2022-04-04:
You're right, SDR# does cut off the edges, I didn't notice that before, thanks.

closed 2022-04-04 by AlexandreRouma