AlexandreRouma/SDRPlusPlus #714: please add KiwiSDR network streaming support

bubnikv, 2022-04-05:
Dear Alexandre.

Would you please add a networked interface to KiwiSDR radios?

KiwiSDR is a linux based multi-channel web based stand-alone networked receiver.
http://kiwisdr.com/

There are many of them around.
http://[kiwisdr.com/public/](http://kiwisdr.com/public/)
http://rx.linkfanel.net/
https://ve3sun.com/KiwiSDR/index.php#open-cookie

I believe the KiwiSDR WebSocket based API works in a similar way to the SpyServer: FFT is streamed as one packet type while a narrow band IQ stream is streamed as another packet type. Example python code is provided here:

https://github.com/jks-prv/kiwiclient

Thanks for considering and 73,
Vojtech OK1IAK

AlexandreRouma, 2022-04-05:
Hello,
SDR++ only supports direct IQ, not IQ+FFT.
Also I do not feel confortable connecting to other people's KiwiSDR and risk crashing them because of an error in my code.

bubnikv, 2022-04-06:
> Also I do not feel confortable connecting to other people's KiwiSDR and risk crashing them because of an error in my code.

We could likely get you one of those radios.

> SDR++ only supports direct IQ, not IQ+FFT.

That is a showstopper wrt. the current KiwiSDR firmware / software. The hardware is capable of wideband IQ, but the current API is not.

I think we can close this issue then.

closed 2022-04-06 by bubnikv

AlexandreRouma, 2022-04-06:
I'll clarify that IQ+FFT mode will likely be supported in the future since it's needed for SDR++ server, but it requires quite a lot of modifications

bubnikv, 2023-02-04:
Hi Alex.

SDRAngel supports kiwisdr API, the source code is available.

https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/kiwisdr

Even with 8kHz or 12kHz bandwidth (depending on bandwidth configuration) as long as the software could tune the VFO when scrolling outside of the waterfall boundaries, having a KiwiSDR support in SDR++ would be great. That's how it works in SDRAngel, however the UI is somehow limited.

> Also I do not feel confortable connecting to other people's KiwiSDR and risk crashing them because of an error in my code.

I don't really think this is an issue. There are other SW tools that connect to Kiwis, thus the KiwiSDR API should be quite resilient.

Thanks for reconsidering and 73,
Vojtech OK1IAK