AlexandreRouma/SDRPlusPlus #718: Installing using DEB attempts to uninstall WineHQ and dependencies

SpaceXCheeseWheel, 2022-04-08:
Attempting to install SDR++ using the deb files on the latest version of Linux Mint attempts to uninstall several packages such as WineHQ, libasound, and libjack:

```
libasound2-plugins:i386
libjack-jackd-2-0
libjack-jackd-2-0:i386
wine-devel
wine-devel-i386:i386
winehq-devel
winetricks
```
I am using `sdrpp_ubuntu_focal_amd64.deb` from the most recent release.

[image]

From what I understand, the source of the problem is `libjack0`. `libjack-jackd2-0` is a dependency of wine, and `libjack0` is one of SDR++s' dependencies. According to apt, `libjack0` conflicts with `libjack-jackd2-0`. Apt tries to uninstall libjack-jackd2-0, and it just works up the chain. I could be completely wrong about this, though. 

I am unsure exactly how to troubleshoot this any further. For anyone trying to replicate this, I am using wine packaged by WineHQ following [this guide](https://wiki.winehq.org/Ubuntu). In other words, I am not using wine from the official ubuntu repositories.

AlexandreRouma, 2022-04-08:
either you got the wrong deb or your distro's not up to date or your graphical package manager is broken.
De b files cannot ask to remove packages. It can only ask to install some

SpaceXCheeseWheel, 2022-04-20:
Reproduced using apt, but seems to be exclusive to my current configuration, was not able to reproduce on another Linux mint system. Closing for now.

RealJTG, 2022-04-29:
> either you got the wrong deb or your distro's not up to date or your graphical package manager is broken. Deb files cannot ask to remove packages. It can only ask to install some

Actually apt [can](https://askubuntu.com/questions/1308117/ubuntu-20-04-installing-wine-has-downgraded-packages-cant-see-new-versions) [ask](https://unix.stackexchange.com/questions/154977/why-is-apt-get-trying-to-remove-certain-packages-when-i-install-wine) [for](https://forum.winehq.org/viewtopic.php?f=8&t=35440) [removal](https://community.mycroft.ai/t/there-should-be-a-warning-that-mycroft-will-uninstall-wine-and-other-software/6995), it isn't the smartest package manager, sadly.

I have the same issue with Ubuntu 20.04, and I have Wine installed from [WineHQ repo](https://wiki.winehq.org/Ubuntu) aswell.  The main issue is librtaudio-dev which depends on libjack-dev, and there are some stuff linked against different versions of libjack in the WineHQ repo.

Aptitude shows some more details on this conflict, but I still have no clue what to do, it seems to be no acceptable solution :(

```
> sudo aptitude install libjack-dev
The following NEW packages will be installed:
  libjack-dev libjack0{ab} 
0 packages upgraded, 2 newly installed, 0 to remove and 26 not upgraded.
Need to get 299 kB of archives. After unpacking 2.730 kB will be used.
The following packages have unmet dependencies:
 libjack-jackd2-0 : Conflicts: libjack-0.116 which is a virtual package, provided by:
                               - libjack0 (1:0.125.0-3build2), but 1:0.125.0-3build2 is to be installed
                               - libjack-jackd2-0 (1.9.12~dfsg-2ubuntu2), but 1.9.12~dfsg-2ubuntu2 is installed

                    Conflicts: libjack-0.125 which is a virtual package, provided by:
                               - libjack0 (1:0.125.0-3build2), but 1:0.125.0-3build2 is to be installed
                               - libjack-jackd2-0 (1.9.12~dfsg-2ubuntu2), but 1.9.12~dfsg-2ubuntu2 is installed

                    Conflicts: libjack0 but 1:0.125.0-3build2 is to be installed
 libjack-jackd2-0:i386 : Conflicts: libjack-0.116 which is a virtual package, provided by:
                                    - libjack0 (1:0.125.0-3build2), but 1:0.125.0-3build2 is to be installed
                                    - libjack-jackd2-0 (1.9.12~dfsg-2ubuntu2), but 1.9.12~dfsg-2ubuntu2 is installed

                         Conflicts: libjack-0.125 which is a virtual package, provided by:
                                    - libjack0 (1:0.125.0-3build2), but 1:0.125.0-3build2 is to be installed
                                    - libjack-jackd2-0 (1.9.12~dfsg-2ubuntu2), but 1.9.12~dfsg-2ubuntu2 is installed

                         Conflicts: libjack0 but 1:0.125.0-3build2 is to be installed
 libjack0 : Conflicts: libjack-0.116 which is a virtual package, provided by:
                       - libjack0 (1:0.125.0-3build2), but 1:0.125.0-3build2 is to be installed
                       - libjack-jackd2-0 (1.9.12~dfsg-2ubuntu2), but 1.9.12~dfsg-2ubuntu2 is installed

            Conflicts: libjack-0.116:i386 which is a virtual package, provided by:
                       - libjack0:i386 (1:0.125.0-3build2), but it is not going to be installed
                       - libjack-jackd2-0:i386 (1.9.12~dfsg-2ubuntu2), but 1.9.12~dfsg-2ubuntu2 is installed

            Conflicts: libjack-0.125 which is a virtual package, provided by:
                       - libjack0 (1:0.125.0-3build2), but 1:0.125.0-3build2 is to be installed
                       - libjack-jackd2-0 (1.9.12~dfsg-2ubuntu2), but 1.9.12~dfsg-2ubuntu2 is installed

            Conflicts: libjack-0.125:i386 which is a virtual package, provided by:
                       - libjack0:i386 (1:0.125.0-3build2), but it is not going to be installed
                       - libjack-jackd2-0:i386 (1.9.12~dfsg-2ubuntu2), but 1.9.12~dfsg-2ubuntu2 is installed

The following actions will resolve these dependencies:

     Remove the following packages:                             
1)     libasound2-plugins:i386 [1.2.2-1ubuntu1 (focal, now)]    
2)     libjack-jackd2-0 [1.9.12~dfsg-2ubuntu2 (focal, now)]     
3)     libjack-jackd2-0:i386 [1.9.12~dfsg-2ubuntu2 (focal, now)]
4)     playonlinux [4.3.4-1ubuntu1 (focal, now)]                
5)     wine-devel [7.7~focal-1 (focal, now)]                    
6)     wine-devel-i386:i386 [7.7~focal-1 (focal, now)]          
7)     winehq-devel [7.7~focal-1 (focal, now)] 
```

SpaceXCheeseWheel, 2022-05-01:
Good to know that I'm not alone with this issue, thought it was just me. I'm not experienced enough to know how to resolve this, so I'll go ahead and re-open this.

AlexandreRouma, 2022-05-01:
> Actually apt [can](https://askubuntu.com/questions/1308117/ubuntu-20-04-installing-wine-has-downgraded-packages-cant-see-new-versions) [...]

A conflict isn't asking for something to be removed and SDR++ doesn't list any conflict. It's `libjack`'s maintainer's fault for not setting it up in such a way that it needs to remove other packages when installed (it's a dependency of rtaudio, I don't ask for it to be installed).

You can check the exact code that generates the deb and confirm that it doesn't ask for such a thing (https://github.com/AlexandreRouma/SDRPlusPlus/blob/master/make_debian_package.sh)

TL;DR There is nothing I can do to fix the incompetence of the Ubuntu/libjack mainteriners. Closing the issue

closed 2022-05-01 by AlexandreRouma

SpaceXCheeseWheel, 2022-05-01:
I see; I'll go ahead and see if it's reasonable to submit an issue to the libjack maintainers, but just looking at it, it's probably not something the maintainers can snap your fingers and fix. For the time being, though, is there any workaround? Not asking you in particular, just anyone who might have an answer. Wine is a fairly frequently installed program, so I'm kind of surprised this issue hasn't come up sooner.

RealJTG, 2022-05-02:
Possible workaround (works for me in Ubuntu 20.04.4 with `https://dl.winehq.org/wine-builds/ubuntu/ focal main` repo in sources.list):

```
# This somehow provides libjack-dev without further conflicts :)
sudo apt-get install libjack-jackd2-dev

# librtaudio-dev seems to be happy now
sudo apt-get install librtaudio-dev

# Install sdrpp
sudo dpkg -i sdrpp_ubuntu_focal_amd64.deb

# Install missing dependencies
sudo apt --fix-broken install 
```

PilotBob42, 2023-03-05:
> Possible workaround (works for me in Ubuntu 20.04.4 with `https://dl.winehq.org/wine-builds/ubuntu/ focal main` repo in sources.list):

Thank you for this. This workaround did the trick for me  on Linux Mint 21.1 (Ubuntu 22.04.2 "Jammy" based).

xproot, 2023-03-09:
> Possible workaround (works for me in Ubuntu 20.04.4 with `https://dl.winehq.org/wine-builds/ubuntu/ focal main` repo in sources.list):
> 
> ```
> # This somehow provides libjack-dev without further conflicts :)
> sudo apt-get install libjack-jackd2-dev
> 
> # librtaudio-dev seems to be happy now
> sudo apt-get install librtaudio-dev
> 
> # Install sdrpp
> sudo dpkg -i sdrpp_ubuntu_focal_amd64.deb
> 
> # Install missing dependencies
> sudo apt --fix-broken install 
> 
> sdrpp
> [2022-05-02 18:19:14.545] [info] SDR++ v1.0.4
> ...
> [2022-05-02 18:19:15.758] [info] Ready.
> ^C
> 
> wine --version
> wine-7.7
> ```
> 
> \o/

This workaround works on Debian 11 (bullseye) too, thanks.