AlexandreRouma/SDRPlusPlus #723: Study the SDRPlusPlus Code Without Installing

KGx55, 2022-04-10:
This is my first post and I hope it is ok.  I am hoping to study the Mac version of SDRPlusPlus to learn from it before attempting to build it.  I downloaded sdrpp_macos_amd64.pkg and unpacked it with the "pkgutil --expand" utility.  The result had a file titled Sdrpp.pkg, which I viewed with “show package contents.”  There, I could see three files.  Two of those files, Payload and Bom were encoded and I was not able to read them.  Is there a way to read the whole project code without installing the package?

gvanem, 2022-04-11:
Why don't you download the sources instead:
`git clone https://github.com/AlexandreRouma/SDRPlusPlus.git <some_empty_directory>`

Or download the `.zip` from the frontpage.

shuyuan-liu, 2022-04-11:
You could also read the code here online, for example https://github.com/AlexandreRouma/SDRPlusPlus/blob/master/core/src/gui/main_window.cpp.

closed 2022-08-17 by AlexandreRouma