AlexandreRouma/SDRPlusPlus #730: ./codec2/version.h Not found

farawaysouthwest, 2022-04-16:
Hi All,

I'm building an Apple M1 image and I get the following error when building from source. 
`... /homebrew/Cellar/codec2/1.0.3/include/codec2/codec2.h:32:10: fatal error: 'codec2/version.h' file not found`

I installed `codec2` via homebrew, but it seems to be an incompatible version? Has anyone else experienced this?

gvanem, 2022-04-27:
The `<codec2_root>/include/codec2/version.h` is supposed to be generated from `<codec2_root>/cmake/version.ih.n` when building.  But it's a small file that you can add with only this:
```c
#ifndef CODEC2_VERSION_H
#define CODEC2_VERSION_H

#define CODEC2_VERSION_MAJOR  1
#define CODEC2_VERSION_MINOR  0
#define CODEC2_VERSION_PATCH  0
#define CODEC2_VERSION       "1.0.0"
#endif
```

packetpilot, 2022-06-24:
I worked around this by editing this file:
/opt/homebrew/Cellar/codec2/1.0.3/include/codec2/codec2.h

...as so:
```
/*#include <codec2/version.h>*/
#include <version.h>
```
...such that the now-commented line is the original (L32), and the subsequent is what allowed the build to succeed.

Without the lensing to open a PR myself (nor the time to invest to acquire it), perhaps the above would help @gvanem turn his suggested workaround into a proper fix. 🤞

AlexandreRouma, 2022-09-13:
This should probably be brought up to the attention of the codec2 developpers. I load it like any other library, if it doesn't install itself properly it should not be fixed in SDR++ but instead at the library.

closed 2022-09-13 by AlexandreRouma