AlexandreRouma/SDRPlusPlus #732: sdr++ tcp server win11

khanfar, 2022-04-19:
i try using terminal cmd in windows 11 to start sdr++ server , i try this command : sdrpp.exe -—server -s -p 5500

i got this error : 
SDR++ v1.0.4
[2022-04-19 20:32:52.498] [error] Invalid command line option: -p

please help , whats wrong in my command ?

i try to run my rsp as a server in pc and use android version of sdr++ to receive via local wifi lan , please help

AlexandreRouma, 2022-04-19:
you're running a 6 month old version and trying to use a feature that was implemented only two months ago.
Download the nightly: https://www.sdrpp.org/nightly

shuyuan-liu, 2022-04-19:
I'll add that in the current version `-s` and `--server` mean the same, and there is no longer a command line option to show logs as far as I know. You may want to remove one of them from your command, although keeping both won't do any harm.

AlexandreRouma, 2022-04-19:
There is, but in server mode it's now enabled by zero (it was a bug that it wasn't)
the new option name is `-c` to show the console

aosync, 2022-08-17:
Issue staled after suggestion. Feel free to reopen.

closed 2022-08-17 by aosync