AlexandreRouma/SDRPlusPlus #733: 100% CPU utilisation while idle

WayneBooth, 2022-04-21:
AMD Ryzen 7 3700X Processor
32 Gb Memory
Ubuntu Focal

I've recently upgraded to nightly build for the "SDR++ Server" functionality. However I've noticed persistent audio choppiness. When investigating I see the CPU utilisation of SDR++ is close to 100% (on a single CPU) while being idle (even after startup and before connecting to any source). Nothing looks out of the ordinary, although I do notice the framerate (from the debug tab) is hovering around the 10000's mark.

Any thoughts?

WayneBooth, 2022-04-21:
Apologies, framerate is around 1000 (not 10000), my mistake.
I tried starting with __GL_SYNC_TO_VBLANK=0 (since I'm using Nvidia) but it made no difference (neither did vblank_mode=1, checked for good measure)

AlexandreRouma, 2022-04-21:
Sounds like a broken graphics driver or something interfering with it. VSync is hardcoded to be on all the time.

parkerlreed, 2022-04-21:
> Thanks for your reply. Apologies, framerate is around 1000 (not 10000), my mistake. I tried starting with __GL_SYNC_TO_VBLANK=0 (since I'm using Nvidia) but it made no difference (neither did vblank_mode=1, checked for good measure)

I'm an idiot. Copy and paste mistake. Try it with 1 instead of 0

WayneBooth, 2022-04-21:
Thanks for the feedback.
I've reset my video driver setting/configuration and now things are acting normally. (60FPS)

closed 2022-04-21 by WayneBooth