AlexandreRouma/SDRPlusPlus #734: Assertion failed on master release

parkerlreed, 2022-04-21:
Arch Linux
gcc 11.2.0
pipewire 1:0.3.50-1
pipewire-jack 1:0.3.50-1

Seems to be failing with pipewire as the jack server.

Backtrace https://gist.github.com/86c946252d22874ad48700514ca355c1

```
[parker@t495 sdrpp-git]$ sdrpp
[2022-04-21 12:54:15.584] [info] SDR++ v1.0.6
[2022-04-21 12:54:15.584] [info] Loading config
[2022-04-21 12:54:15.585] [warning] ConfigManager locked, waiting...
[2022-04-21 12:54:16.231] [info] Using OpenGL 3.0
[2022-04-21 12:54:16.341] [info] Loading icons
[2022-04-21 12:54:16.351] [info] Loading band plans
[2022-04-21 12:54:16.354] [info] Loading band plans color table
[2022-04-21 12:54:16.359] [info] Loading modules
[2022-04-21 12:54:16.359] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2022-04-21 12:54:16.360] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-04-21 12:54:16.362] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2022-04-21 12:54:16.363] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2022-04-21 12:54:16.365] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2022-04-21 12:54:16.366] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2022-04-21 12:54:16.368] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2022-04-21 12:54:16.371] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2022-04-21 12:54:16.375] [info] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[2022-04-21 12:54:16.377] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2022-04-21 12:54:16.378] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-04-21 12:54:16.380] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-04-21 12:54:16.381] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2022-04-21 12:54:16.384] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2022-04-21 12:54:16.391] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2022-04-21 12:54:16.392] [info] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[2022-04-21 12:54:16.393] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2022-04-21 12:54:16.395] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2022-04-21 12:54:16.396] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2022-04-21 12:54:16.397] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2022-04-21 12:54:16.400] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2022-04-21 12:54:16.401] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2022-04-21 12:54:16.405] [info] Initializing Airspy Source (airspy_source)
[2022-04-21 12:54:16.419] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-04-21 12:54:16.431] [info] Initializing Audio Sink (audio_sink)
[2022-04-21 12:54:16.431] [info] Initializing BladeRF Source (bladerf_source)
[2022-04-21 12:54:16.444] [error] Could not list devices -7
[2022-04-21 12:54:16.444] [info] Initializing File Source (file_source)
[2022-04-21 12:54:16.444] [info] Initializing Frequency Manager (frequency_manager)
[2022-04-21 12:54:16.446] [info] Initializing HackRF Source (hackrf_source)
[2022-04-21 12:54:16.459] [info] Initializing LimeSDR Source (limesdr_source)
[2022-04-21 12:54:16.481] [info] Initializing Network Sink (network_sink)
[2022-04-21 12:54:16.481] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-04-21 12:54:16.482] [info] Initializing RFspace Source (rfspace_source)
[2022-04-21 12:54:16.485] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-04-21 12:54:16.500] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-04-21 12:54:16.502] [info] Initializing Radio (radio)

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

/usr/include/c++/11.2.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = RtAudio::DeviceInfo; _Alloc = std::allocator<RtAudio::DeviceInfo>; std::vector<_Tp, _Alloc>::reference = RtAudio::DeviceInfo&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Aborted (core dumped)
```

AlexandreRouma, 2022-04-21:
install pulse

parkerlreed, 2022-04-21:
pipewire also has pulse compatibility which is installed and enabled. Is SDR++ hardcoded for pulseaudio?

```
[parker@t495 ~]$ pactl info
Server String: /run/user/1000/pulse/native
Library Protocol Version: 35
Server Protocol Version: 35
Is Local: yes
Client Index: 233
Tile Size: 65472
User Name: parker
Host Name: t495
Server Name: PulseAudio (on PipeWire 0.3.50)
Server Version: 15.0.0
Default Sample Specification: float32le 2ch 48000Hz
Default Channel Map: front-left,front-right
Default Sink: alsa_output.pci-0000_06_00.1.hdmi-stereo-extra1
Default Source: alsa_input.pci-0000_06_00.6.analog-stereo
Cookie: 302d:e51e
```

AlexandreRouma, 2022-04-21:
pretty sure RtAudio doesn't support it properly

AlexandreRouma, 2022-04-21:
Also looking at the logs RtAudio seems to insist on using jack instead of pulse. So might wanna follow the instructions to recompile it with pulseaudio as the default:
https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

parkerlreed, 2022-04-21:
That indeed worked! Arch uses meson for the build so the options were a little different 

rtaudio's pulse support is working fine with pipewire. The issue seems to just be the picking JACK first and failing.

```
build() {
  local meson_options=(
    -D core=disabled
    -D dsound=disabled
    -D asio=disabled
    -D wasapi=disabled
    -D docs=true
    -D install_docs=true
    -D jack=disabled
    -D alsa=disabled
  )

  arch-meson ${pkgbase}-${pkgver} build "${meson_options[@]}"
  meson compile -C build
}
```

closed 2022-04-21 by parkerlreed