AlexandreRouma/SDRPlusPlus #747: SDR++ currently not opening on Ubuntu/Pop!_OS 22.04

JoshuaKimsey, 2022-05-01:
When clicking on the app icon to launch, it acts as though it was never clicked. When launched from the terminal using `sdrpp` it responds with the following error: `sdrpp: error while loading shared libraries: libGLEW.so.2.1: cannot open shared object file: No such file or directory`

I searched for the libGlew with `apt search` and found the version shipped with Ubuntu/Pop!_OS is: `libglew2.2/jammy,now 2.2.0-4 amd64`, which seems to be where the error lies, given SDR++ is looking for version 2.1. I tried messing around with the Debian Sid version for fun, but it failed for other reasons, so it seems like the libGLEW dependency needs to be updated to version 2.2.

AlexandreRouma, 2022-05-01:
Duplicate of https://github.com/AlexandreRouma/SDRPlusPlus/issues/745

you installed the wrong deb. Install the one specifically for Jammy from the nightly build: https://www.sdrpp.org/nightly

closed 2022-05-01 by AlexandreRouma

JoshuaKimsey, 2022-05-01:
Gotcha, yeah that worked perfectly! I actually didn't know SDR++ had a website now, cool to see! :)