AlexandreRouma/SDRPlusPlus #748: Debian / Ubuntu installation instructions should mention nightly builds

shuyuan-liu, 2022-05-01:
Issues #745 and #747 might indicate many users aren't aware of nightly builds, and try to install the wrong package when there isn't a release package available for their system. In fact the main README doesn't mention "nightly" anywhere.

I think it would be useful to mention nightly builds under the [install instructions for Ubuntu](https://github.com/AlexandreRouma/SDRPlusPlus#debian-based-ubuntu-mint-etc), either for Jammy users specifically, or something like "if no package matches your OS in Releases, please look in the nightly builds at ...".

JoshuaKimsey, 2022-05-01:
I do agree, I had no clue about there being Nightly Builds (or even a website for SDR++), so it definitely should be mentioned somewhere on the main GitHub page.

JosephTheTank13, 2022-07-12:
wait there's a website now?

shuyuan-liu, 2022-07-12:
Yes https://www.sdrpp.org/

maiestro, 2022-08-12:
Somehow I am too stupid to find the download link for the nightly builds, can someone give me a hint to find the ubuntu jammy .deb package from the github actions page?

Would be great, thank you very much!

shuyuan-liu, 2022-08-12:
> Somehow I am too stupid to find the download link for the nightly builds, can someone give me a hint to find the ubuntu jammy .deb package from the github actions page?
> 
> Would be great, thank you very much!

- Go to the Actions page
- Open the latest build marked successful (green check mark)
- Scroll down to Artifacts i.e. packages built
- Click on the package you need. In your case, `sdrpp_ubuntu_jammy_amd64`.

loewal, 2022-08-12:
And you have to login first. This means, you need to have (made) a Github account.

Loe Walter
Insula 103
6416 BZ  Heerlen
☎️ +31-6-24476963

maiestro, 2022-08-12:
Thanks you guys @shuyuan-liu and @loewal for your support,
I'm logged in, but I cant see any files, just log files :

[image]
[image]
[image]

loewal, 2022-08-12:
I the right portion of the screen you can scroll further down.

maiestro, 2022-08-12:
Oh my god, you are totally right, sorry.

Now i see that too - the scrolling with the mouse was in the build_all.yml area as zoom. I with my small screen thought that would already be the end of the page. Thank you :)

loewal, 2022-10-11:
Sorry, use the scrollbar otherwise scrolling sometimes doesn’t work

mmrein, 2022-11-01:
The "must login" part got me. Did try without it and was wondering why is there file list without posibility to dowload them. Ended up building from source for Jammy...

shuyuan-liu, 2022-11-15:
Closing since the readme now mentions nightly builds so the original issue has been resolved.

closed 2022-11-15 by shuyuan-liu