AlexandreRouma/SDRPlusPlus #749: Crash on startup on Fedora/QubesOS [1.0.4]

kord1e, 2022-05-01:
I'm trying to use SDR++ on QubesOS in a Fedora 34 VM. It crashes on startup. Console output is below. I think the crash may be related to audio interfaces.

Please let me know if there is something I could do to help debugging this.

`[2022-05-01 21:51:22.148] [info] SDR++ v1.0.4
[2022-05-01 21:51:22.148] [info] Loading config
[2022-05-01 21:51:22.404] [info] Using OpenGL 3.0
[2022-05-01 21:51:22.656] [info] Loading icons
[2022-05-01 21:51:22.871] [info] Loading band plans
[2022-05-01 21:51:22.882] [info] Loading band plans color table
[2022-05-01 21:51:22.897] [info] Loading modules
[2022-05-01 21:51:22.898] [info] Loading /usr/lib64/sdrpp/plugins/weather_sat_decoder.so
[2022-05-01 21:51:22.909] [info] Loading /usr/lib64/sdrpp/plugins/rtl_sdr_source.so
[2022-05-01 21:51:22.921] [info] Loading /usr/lib64/sdrpp/plugins/radio.so
[2022-05-01 21:51:22.929] [info] Loading /usr/lib64/sdrpp/plugins/discord_integration.so
[2022-05-01 21:51:22.936] [info] Loading /usr/lib64/sdrpp/plugins/meteor_demodulator.so
[2022-05-01 21:51:22.944] [info] Loading /usr/lib64/sdrpp/plugins/rigctl_server.so
[2022-05-01 21:51:22.951] [info] Loading /usr/lib64/sdrpp/plugins/m17_decoder.so
[2022-05-01 21:51:22.973] [info] Loading /usr/lib64/sdrpp/plugins/network_sink.so
[2022-05-01 21:51:22.981] [info] Loading /usr/lib64/sdrpp/plugins/rtl_tcp_source.so
[2022-05-01 21:51:22.987] [info] Loading /usr/lib64/sdrpp/plugins/audio_sink.so
[2022-05-01 21:51:23.019] [info] Loading /usr/lib64/sdrpp/plugins/frequency_manager.so
[2022-05-01 21:51:23.035] [info] Loading /usr/lib64/sdrpp/plugins/hackrf_source.so
[2022-05-01 21:51:23.041] [info] Loading /usr/lib64/sdrpp/plugins/new_portaudio_sink.so
[2022-05-01 21:51:23.067] [info] Loading /usr/lib64/sdrpp/plugins/spyserver_source.so
[2022-05-01 21:51:23.086] [info] Loading /usr/lib64/sdrpp/plugins/soapy_source.so
[2022-05-01 21:51:23.109] [info] Loading /usr/lib64/sdrpp/plugins/recorder.so
[2022-05-01 21:51:23.121] [info] Loading /usr/lib64/sdrpp/plugins/file_source.so
[2022-05-01 21:51:23.140] [info] Initializing Airspy Source (airspy_source)
[2022-05-01 21:51:23.142] [error] Module 'airspy_source' doesn't exist
[2022-05-01 21:51:23.142] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-05-01 21:51:23.152] [error] Module 'airspyhf_source' doesn't exist
[2022-05-01 21:51:23.152] [info] Initializing Audio Sink (audio_sink)
[2022-05-01 21:51:23.161] [info] Initializing BladeRF Source (bladerf_source)
[2022-05-01 21:51:23.165] [error] Module 'bladerf_source' doesn't exist
[2022-05-01 21:51:23.165] [info] Initializing File Source (file_source)
[2022-05-01 21:51:23.168] [info] Initializing Frequency Manager (frequency_manager)
[2022-05-01 21:51:23.172] [info] Initializing HackRF Source (hackrf_source)
[2022-05-01 21:51:23.176] [info] Initializing LimeSDR Source (limesdr_source)
[2022-05-01 21:51:23.179] [error] Module 'limesdr_source' doesn't exist
[2022-05-01 21:51:23.180] [info] Initializing Network Sink (network_sink)
[2022-05-01 21:51:23.184] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-05-01 21:51:23.187] [error] Module 'plutosdr_source' doesn't exist
[2022-05-01 21:51:23.187] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-05-01 21:51:23.190] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-05-01 21:51:23.192] [info] Initializing Radio (radio)
[2022-05-01 21:51:23.270] [error] Could not open audio device
[2022-05-01 21:51:23.270] [info] Initializing Recorder (recorder)
[2022-05-01 21:51:23.273] [info] Initializing Rigctl Server (rigctl_server)
[2022-05-01 21:51:23.277] [info] Initializing SDRplay Source (sdrplay_source)
[2022-05-01 21:51:23.281] [error] Module 'sdrplay_source' doesn't exist
[2022-05-01 21:51:23.281] [info] Initializing SoapySDR Source (soapy_source)
[2022-05-01 21:51:23.300] [info] Initializing SpyServer Source (spyserver_source)
[2022-05-01 21:51:23.331] [info] Loading color maps
[2022-05-01 21:51:23.349] [info] Loading /usr/share/sdrpp/colormaps/turbo.json
[2022-05-01 21:51:23.355] [info] Loading /usr/share/sdrpp/colormaps/gqrx.json
[2022-05-01 21:51:23.361] [info] Loading /usr/share/sdrpp/colormaps/plasma.json
[2022-05-01 21:51:23.369] [info] Loading /usr/share/sdrpp/colormaps/greyscale.json
[2022-05-01 21:51:23.373] [info] Loading /usr/share/sdrpp/colormaps/classic.json
[2022-05-01 21:51:23.377] [info] Loading /usr/share/sdrpp/colormaps/magma.json
[2022-05-01 21:51:23.388] [info] Loading /usr/share/sdrpp/colormaps/inferno.json
[2022-05-01 21:51:23.392] [info] Loading /usr/share/sdrpp/colormaps/electric.json
[2022-05-01 21:51:23.398] [info] Loading /usr/share/sdrpp/colormaps/viridis.json
[2022-05-01 21:51:23.401] [info] Loading /usr/share/sdrpp/colormaps/websdr.json
[2022-05-01 21:51:23.405] [info] Loading /usr/share/sdrpp/colormaps/classic_green.json
[2022-05-01 21:51:23.425] [info] New DSP samplerate: 1000000 (source samplerate is 1000000)
[2022-05-01 21:51:23.430] [info] Updating FFT DSP settings: Keep: 50000, Skip: 0
[2022-05-01 21:51:23.431] [info] FileSourceModule 'File Source': Tune: 100000000!
[2022-05-01 21:51:23.431] [info] FileSourceModule 'File Source': Menu Select!
[2022-05-01 21:51:23.431] [info] New DSP samplerate: 1000000 (source samplerate is 1000000)
[2022-05-01 21:51:23.431] [info] Updating FFT DSP settings: Keep: 50000, Skip: 0
terminate called after throwing an instance of 'RtAudioError'
  what():  RtApi:: a stream is not open!
Aborted (core dumped)
`

SamRH, 2022-05-04:
This probably relates to https://github.com/AlexandreRouma/SDRPlusPlus/pull/679/files and https://github.com/thestk/rtaudio/pull/344

With the latest version of pipewire it's probably easiest to start SDR++ with jack support disabled as PIPEWIRE_PROPS doesn't seem to be working as documented (known issue AFAIK, will be fixed in the next release of pipewire). You may need to delete ~/.config/sdrpp/audio_sink_config.json (or equivalent depending on how you've installed sdrpp) before the crashes stop.

Example commands:
```
rm ~/.config/sdrpp/audio_sink_config.json 
env PIPEWIRE_NOJACK=1 sdrpp
```

kord1e, 2022-05-07:
> Example commands:
> 
> ```
> rm ~/.config/sdrpp/audio_sink_config.json 
> env PIPEWIRE_NOJACK=1 sdrpp
> ```

This prevented the crash. But SDR++ hangs when it tries to load some colormaps. It's loading "classic_green.json and the following is printed to the console. Or if the classic_green.json file is removed, it hangs when loading "websdr.json".

[2022-05-07 15:50:12.061] [info] FileSourceModule 'File Source': Tune: 100000000!
[2022-05-07 15:50:12.061] [info] FileSourceModule 'File Source': Menu Select!
[2022-05-07 15:50:12.061] [info] New DSP samplerate: 1000000 (source samplerate is 1000000)
[2022-05-07 15:50:12.061] [info] Updating FFT DSP settings: Keep: 50000, Skip: 0
[2022-05-07 15:50:12.209] [warning] ConfigManager locked, waiting...
[2022-05-07 15:50:13.209] [warning] ConfigManager locked, waiting...
[2022-05-07 15:50:14.209] [warning] ConfigManager locked, waiting...
[2022-05-07 15:50:15.209] [warning] ConfigManager locked, waiting...
[2022-05-07 15:50:16.209] [warning] ConfigManager locked, waiting...

kord1e, 2022-05-07:
Full console output here:

$ PIPEWIRE_NOJACK=1 sdrpp
[2022-05-07 16:10:04.991] [info] SDR++ v1.0.4
[2022-05-07 16:10:04.991] [info] Loading config
[2022-05-07 16:10:05.209] [info] Using OpenGL 3.0
[2022-05-07 16:10:05.370] [info] Loading icons
[2022-05-07 16:10:05.390] [info] Loading band plans
[2022-05-07 16:10:05.394] [info] Loading band plans color table
[2022-05-07 16:10:05.404] [info] Loading modules
[2022-05-07 16:10:05.405] [info] Loading /usr/lib64/sdrpp/plugins/weather_sat_decoder.so
[2022-05-07 16:10:05.408] [info] Loading /usr/lib64/sdrpp/plugins/rtl_sdr_source.so
[2022-05-07 16:10:05.412] [info] Loading /usr/lib64/sdrpp/plugins/radio.so
[2022-05-07 16:10:05.416] [info] Loading /usr/lib64/sdrpp/plugins/discord_integration.so
[2022-05-07 16:10:05.419] [info] Loading /usr/lib64/sdrpp/plugins/meteor_demodulator.so
[2022-05-07 16:10:05.423] [info] Loading /usr/lib64/sdrpp/plugins/rigctl_server.so
[2022-05-07 16:10:05.426] [info] Loading /usr/lib64/sdrpp/plugins/m17_decoder.so
[2022-05-07 16:10:05.432] [info] Loading /usr/lib64/sdrpp/plugins/network_sink.so
[2022-05-07 16:10:05.435] [info] Loading /usr/lib64/sdrpp/plugins/rtl_tcp_source.so
[2022-05-07 16:10:05.441] [info] Loading /usr/lib64/sdrpp/plugins/audio_sink.so
[2022-05-07 16:10:05.453] [info] Loading /usr/lib64/sdrpp/plugins/frequency_manager.so
[2022-05-07 16:10:05.458] [info] Loading /usr/lib64/sdrpp/plugins/hackrf_source.so
[2022-05-07 16:10:05.463] [info] Loading /usr/lib64/sdrpp/plugins/new_portaudio_sink.so
[2022-05-07 16:10:05.469] [info] Loading /usr/lib64/sdrpp/plugins/spyserver_source.so
[2022-05-07 16:10:05.472] [info] Loading /usr/lib64/sdrpp/plugins/soapy_source.so
[2022-05-07 16:10:05.475] [info] Loading /usr/lib64/sdrpp/plugins/recorder.so
[2022-05-07 16:10:05.479] [info] Loading /usr/lib64/sdrpp/plugins/file_source.so
[2022-05-07 16:10:05.482] [info] Initializing Airspy Source (airspy_source)
[2022-05-07 16:10:05.484] [error] Module 'airspy_source' doesn't exist
[2022-05-07 16:10:05.484] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-05-07 16:10:05.488] [error] Module 'airspyhf_source' doesn't exist
[2022-05-07 16:10:05.488] [info] Initializing Audio Sink (audio_sink)
[2022-05-07 16:10:05.495] [info] Initializing BladeRF Source (bladerf_source)
[2022-05-07 16:10:05.498] [error] Module 'bladerf_source' doesn't exist
[2022-05-07 16:10:05.498] [info] Initializing File Source (file_source)
[2022-05-07 16:10:05.501] [info] Initializing Frequency Manager (frequency_manager)
[2022-05-07 16:10:05.504] [info] Initializing HackRF Source (hackrf_source)
[2022-05-07 16:10:05.509] [info] Initializing LimeSDR Source (limesdr_source)
[2022-05-07 16:10:05.512] [error] Module 'limesdr_source' doesn't exist
[2022-05-07 16:10:05.512] [info] Initializing Network Sink (network_sink)
[2022-05-07 16:10:05.514] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-05-07 16:10:05.517] [error] Module 'plutosdr_source' doesn't exist
[2022-05-07 16:10:05.517] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
[2022-05-07 16:10:06.173] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-05-07 16:10:06.177] [info] Initializing Radio (radio)
[2022-05-07 16:10:06.356] [info] RtAudio stream open
[2022-05-07 16:10:06.356] [info] Initializing Recorder (recorder)
[2022-05-07 16:10:06.372] [info] Initializing Rigctl Server (rigctl_server)
[2022-05-07 16:10:06.374] [info] Initializing SDRplay Source (sdrplay_source)
[2022-05-07 16:10:06.378] [error] Module 'sdrplay_source' doesn't exist
[2022-05-07 16:10:06.378] [info] Initializing SoapySDR Source (soapy_source)
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Detached kernel driver
Found Rafael Micro R820T tuner
Reattached kernel driver
[2022-05-07 16:10:07.816] [info] Setting sample rate to 250000
[2022-05-07 16:10:07.816] [info] New DSP samplerate: 250000 (source samplerate is 250000)
[2022-05-07 16:10:07.816] [info] Updating FFT DSP settings: Keep: 12500, Skip: 0
[2022-05-07 16:10:07.816] [info] Initializing SpyServer Source (spyserver_source)
[2022-05-07 16:10:07.824] [info] Loading color maps
[2022-05-07 16:10:07.833] [info] Loading /usr/share/sdrpp/colormaps/turbo.json
[2022-05-07 16:10:07.837] [info] Loading /usr/share/sdrpp/colormaps/gqrx.json
[2022-05-07 16:10:07.841] [info] Loading /usr/share/sdrpp/colormaps/plasma.json
[2022-05-07 16:10:07.855] [info] Loading /usr/share/sdrpp/colormaps/greyscale.json
[2022-05-07 16:10:07.859] [info] Loading /usr/share/sdrpp/colormaps/classic.json
[2022-05-07 16:10:07.862] [info] Loading /usr/share/sdrpp/colormaps/magma.json
[2022-05-07 16:10:07.865] [info] Loading /usr/share/sdrpp/colormaps/inferno.json
[2022-05-07 16:10:07.868] [info] Loading /usr/share/sdrpp/colormaps/electric.json
[2022-05-07 16:10:07.870] [info] Loading /usr/share/sdrpp/colormaps/viridis.json
[2022-05-07 16:10:07.873] [info] Loading /usr/share/sdrpp/colormaps/websdr.json
[2022-05-07 16:10:07.880] [info] Loading /usr/share/sdrpp/colormaps/classic_green.json
[2022-05-07 16:10:07.901] [info] New DSP samplerate: 1000000 (source samplerate is 1000000)
[2022-05-07 16:10:07.901] [info] Updating FFT DSP settings: Keep: 50000, Skip: 0
[2022-05-07 16:10:07.902] [info] FileSourceModule 'File Source': Tune: 100000000!
[2022-05-07 16:10:07.902] [info] FileSourceModule 'File Source': Menu Select!
[2022-05-07 16:10:07.902] [info] New DSP samplerate: 1000000 (source samplerate is 1000000)
[2022-05-07 16:10:07.902] [info] Updating FFT DSP settings: Keep: 50000, Skip: 0
[2022-05-07 16:10:07.992] [warning] ConfigManager locked, waiting...
[2022-05-07 16:10:08.992] [warning] ConfigManager locked, waiting...
[2022-05-07 16:10:09.993] [warning] ConfigManager locked, waiting...
[2022-05-07 16:10:10.993] [warning] ConfigManager locked, waiting...
[2022-05-07 16:10:11.993] [warning] ConfigManager locked, waiting...
^C

SamRH, 2022-05-07:
There might have been other config files that needed to be reset. Could try making a backup of the config directory (if you have any bookmarks etc), deleting it, and then trying to start sdrpp with PIPEWIRE_NOJACK set.

kord1e, 2022-05-20:
> There might have been other config files that needed to be reset. Could try making a backup of the config directory (if you have any bookmarks etc), deleting it, and then trying to start sdrpp with PIPEWIRE_NOJACK set.

Even with the entire config directory deleted (~/.config/sdrpp), it still hangs and keeps printing those "ConfigManager locked, waiting" messages.

kord1e, 2022-06-04:
Seems to be working now, even without PIPEWIRE_NOJACK, so I'm closing this.

closed 2022-06-04 by kord1e