AlexandreRouma/SDRPlusPlus #750: The frequency tabs...

kmp56, 2022-05-02:
The frequency tabs overlap and obscure each other if the frequencies of each are close. Maybe the bookmarks bar should be on 2 or more levels (> 1 lines). Is it possible ?
[image]

shuyuan-liu, 2022-05-02:
This has been asked in #211; no updates from the developer yet. GQRX staggers the labels so they don't overlap, but SDR++ doesn't have this feature.

darauble, 2022-06-23:
I have copied Frequency Manager and created my own plugin for exactly this issue:

https://github.com/darauble/bookmark_manager

[image]

AlexandreRouma, 2022-06-23:
Don't post in ever goddamn issue to advertise your plugin