AlexandreRouma/SDRPlusPlus #751: [cosmetic] white icons have low visibility with the light theme

shuyuan-liu, 2022-05-02:
Using the light theme, these white button icons are difficult to see on the light background:

[image]

Although they look OK in isolation, they have much lower contrast than in the dark themes. It also feels odd to have white button icons when all other text is black.

The icons are currently PNGs with fixed colours. Would it be more flexible to implement them as vector graphics with colours specified by themes (either seperately or simply the same as text colour)?

OttoPattemore, 2022-05-03:
ImGui::ImageButtons have a tint argument which can be used to color an image.
It multiples every pixel by the tint.

Here we are using the current theme text color as the tint ( `` ImGui::GetStyleColorVec4(ImGuiCol_Text) ``).
```c++
ImGui::ImageButton(icons::PLAY, btnSize, ImVec2(0, 0), ImVec2(1, 1), 5,ImVec4(0,0,0,0),ImGui::GetStyleColorVec4(ImGuiCol_Text))
```
Modified line 382 in main_window.cpp

In light mode:
[image]

In dark mode:
[image]

shuyuan-liu, 2022-05-03:
This looks quite good. @AlexandreRouma would you like to implement it this way?

AlexandreRouma, 2022-05-10:
I'll try to implement this if I get some free time, extremely busy at the moment and can't go over pull requests

closed 2022-06-21 by AlexandreRouma