AlexandreRouma/SDRPlusPlus #754: Android compilation: Missing libraries

bubnikv, 2022-05-10:
Hello Alex.

CMake for Android native module is not finished. Namely the https://github.com/gnuradio/volk needs to be precompiled and it is not documented how. Would you please polish and document the Android build?

Also it would be great, if you could please support compilation without soapysdr dependency. I believe you already did that for your Android build, thus it would be great to support that on desktop builds as well. I am looking forward to extend sdr++ for sound card based SDR receivers / transceivers and it would make my life much easier.

Thanks and 73, 
Vojtech OK1IAK

ericek111, 2022-05-10:
Vojto, you can easily disable the SoapySDR module by passing `-DOPT_BUILD_SOAPY_SOURCE=OFF` into CMake.

bubnikv, 2022-05-10:
> Vojto, you can easily disable the SoapySDR module by passing
-DOPT_BUILD_SOAPY_SOURCE=OFF into CMake.

Thanks, I think I tried that, however sdr++ still requires libvolk, which I
failed to compile on Windows.
I am not at that computer as of now, I have to try once more.

AlexandreRouma, 2022-05-10:
Android compilation is not supported hence why it's not explained in the readme.
Special modified versions of the libraries are required and I am not ready to release them until they are stable.

closed 2022-05-10 by AlexandreRouma