AlexandreRouma/SDRPlusPlus #756: Running CMAKE Results in pkg_config error (BUG) MAC OSX

ChrisDev8, 2022-05-11:
# When running CMAKE command as stated in the documentation, it creates this error multiple times with different modules.
# OS: MAC M1 OSX; Version: MacOS Monterey 12.1

## Short Error:
```
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
    Reason given by package: The command
      "/usr/lib/pkgconfig" --version
    failed with output:

    stderr: 
      
    result: 
Permission denied
```

## Complete Error:

https://pastebin.com/Yp8unJeR
**Password**: d12FSfZ0aQ

ChrisDev8, 2022-05-11:
Tried sudo prefix
Everything else too complicated.
Also I've had really bad luck using cmake for ANY other programs.

ChrisDev8, 2022-05-12:
# I GOT IT; YES YES YES

## SHORT: https://bit.ly/3wp8k1t

## RAW: https://nightly.link/AlexandreRouma/SDRPlusPlus/workflows/build_all.yaml/master/sdrpp_macos_intel.zip

closed 2022-05-12 by ChrisDev8

AlexandreRouma, 2022-05-12:
see https://github.com/AlexandreRouma/SDRPlusPlus/discussions/757#discussioncomment-2740822 for instructions on how to get the MacOS nightly.