AlexandreRouma/SDRPlusPlus #758: cmake error (macos)

ah1102, 2022-05-15:
Build error. Package `brew install glfw` installed

_macos Catalina 10.15.7
intel x86_64_

```
Apple@iMac-Apple build % cmake .. \        
  -DOPT_BUILD_AUDIO_SINK=OFF \
  -DOPT_BUILD_BLADERF_SOURCE=OFF \
  -DOPT_BUILD_M17_DECODER=ON \
  -DOPT_BUILD_NEW_PORTAUDIO_SINK=ON \
  -DOPT_BUILD_PLUTOSDR_SOURCE=OFF \
  -DOPT_BUILD_PORTAUDIO_SINK=ON \
  -DOPT_BUILD_SOAPY_SOURCE=OFF
CMake Deprecation Warning at core/libcorrect/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Checking for module 'glfw3'
--   No package 'glfw3' found
CMake Error at /opt/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:556 (message):
  A required package was not found
Call Stack (most recent call first):
  /opt/local/share/cmake-3.19/Modules/FindPkgConfig.cmake:744 (_pkg_check_modules_internal)
  core/CMakeLists.txt:59 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
```

AlexandreRouma, 2022-09-29:
probably a bad environment, if pkgconfig can't find it, there's nothing I can do about it.

closed 2022-09-29 by AlexandreRouma

peads, 2022-12-29:
@ah1102 @AlexandreRouma Interestingly, I just ran into this same issue, and the workaround is relatively simple (if you're compiling package from source already anyway): 1. build and install glfw from their repo (glfw/glfw), 2. add `export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/usr/local/lib/pkgconfig` temporarily, or permanently as you like, 3. (optional) run `pkg-config --cflags glfw3` to verify it sees the package.