AlexandreRouma/SDRPlusPlus #760: Windows 10 + HackRF One crash

diggit, 2022-05-18:
Hi,
I am trying to use SDR++ on W10 machine with HackRF One.
SW info:
OS: Windows 10 Enterprise, 21H2
SDR++: 1.0.4 official release

When I click on play button, Sdr++ crashes.
Same thing happens, when I click on "refresh button" bandwidth config.
HackRF One is recognized in device manager and is using WinUSB driver.

Sdr++ terminal output:
```
[2022-05-18 10:07:05.594] [info] SDR++ v1.0.4
[2022-05-18 10:07:05.600] [info] Loading config
[2022-05-18 10:07:06.845] [info] Using OpenGL 3.0
[2022-05-18 10:07:06.952] [info] Loading icons
[2022-05-18 10:07:07.021] [info] Loading band plans
[2022-05-18 10:07:07.030] [info] Loading band plans color table
[2022-05-18 10:07:07.036] [info] Loading modules
[2022-05-18 10:07:07.036] [info] Loading ./modules/airspyhf_source.dll
[2022-05-18 10:07:07.056] [info] Loading ./modules/airspy_source.dll
[2022-05-18 10:07:07.068] [info] Loading ./modules/audio_sink.dll
[2022-05-18 10:07:07.083] [info] Loading ./modules/bladerf_source.dll
[2022-05-18 10:07:07.095] [info] Loading ./modules/discord_integration.dll
[2022-05-18 10:07:07.102] [info] Loading ./modules/file_source.dll
[2022-05-18 10:07:07.108] [info] Loading ./modules/frequency_manager.dll
[2022-05-18 10:07:07.117] [info] Loading ./modules/hackrf_source.dll
[2022-05-18 10:07:07.128] [info] Loading ./modules/limesdr_source.dll
[2022-05-18 10:07:07.142] [info] Loading ./modules/m17_decoder.dll
[2022-05-18 10:07:07.156] [info] Loading ./modules/meteor_demodulator.dll
[2022-05-18 10:07:07.163] [info] Loading ./modules/network_sink.dll
[2022-05-18 10:07:07.171] [info] Loading ./modules/plutosdr_source.dll
[2022-05-18 10:07:07.185] [info] Loading ./modules/radio.dll
[2022-05-18 10:07:07.192] [info] Loading ./modules/recorder.dll
[2022-05-18 10:07:07.203] [info] Loading ./modules/rigctl_server.dll
[2022-05-18 10:07:07.210] [info] Loading ./modules/rtl_sdr_source.dll
[2022-05-18 10:07:07.228] [info] Loading ./modules/rtl_tcp_source.dll
[2022-05-18 10:07:07.239] [info] Loading ./modules/sdrplay_source.dll
[2022-05-18 10:07:07.258] [info] Loading ./modules/soapy_source.dll
[2022-05-18 10:07:07.271] [info] Loading ./modules/spyserver_source.dll
[2022-05-18 10:07:07.278] [info] Initializing Airspy Source (airspy_source)
[2022-05-18 10:07:07.331] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-05-18 10:07:07.379] [info] Initializing Audio Sink (audio_sink)
[2022-05-18 10:07:07.381] [info] Initializing BladeRF Source (bladerf_source)
[2022-05-18 10:07:07.428] [error] Could not list devices -7
[2022-05-18 10:07:07.428] [info] Initializing File Source (file_source)
[2022-05-18 10:07:07.430] [info] Initializing Frequency Manager (frequency_manager)
[2022-05-18 10:07:07.432] [info] Initializing HackRF Source (hackrf_source)
[2022-05-18 10:07:07.476] [info] Initializing LimeSDR Source (limesdr_source)
[2022-05-18 10:07:07.478] [info] Initializing Network Sink (network_sink)
[2022-05-18 10:07:07.479] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-05-18 10:07:07.481] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-05-18 10:07:07.527] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-05-18 10:07:07.530] [info] Initializing Radio (radio)
[2022-05-18 10:07:07.612] [info] RtAudio stream open
[2022-05-18 10:07:07.613] [info] Initializing Recorder (recorder)
[2022-05-18 10:07:07.615] [info] Initializing Rigctl Server (rigctl_server)
[2022-05-18 10:07:07.615] [info] Initializing SDRplay Source (sdrplay_source)
[2022-05-18 10:07:07.616] [error] Could not intiatialized the SDRplay API. Make sure that the service is running.
[2022-05-18 10:07:07.616] [info] Initializing SoapySDR Source (soapy_source)
[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail
[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_107500; UHD_4.0.0.HEAD-0-g90ce6062
[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed
[2022-05-18 10:07:10.483] [info] Setting sample rate to 4000.0
[2022-05-18 10:07:10.483] [info] New DSP samplerate: 4000.0 (source samplerate is 4000.0)
[2022-05-18 10:07:10.485] [info] Updating FFT DSP settings: Keep: 200, Skip: 0
[2022-05-18 10:07:10.485] [info] Initializing SpyServer Source (spyserver_source)
[2022-05-18 10:07:10.486] [info] Loading color maps
[2022-05-18 10:07:10.487] [info] Loading ./res/colormaps/classic.json
[2022-05-18 10:07:10.504] [info] Loading ./res/colormaps/classic_green.json
[2022-05-18 10:07:10.512] [info] Loading ./res/colormaps/electric.json
[2022-05-18 10:07:10.522] [info] Loading ./res/colormaps/gqrx.json
[2022-05-18 10:07:10.530] [info] Loading ./res/colormaps/greyscale.json
[2022-05-18 10:07:10.539] [info] Loading ./res/colormaps/inferno.json
[2022-05-18 10:07:10.547] [info] Loading ./res/colormaps/magma.json
[2022-05-18 10:07:10.557] [info] Loading ./res/colormaps/plasma.json
[2022-05-18 10:07:10.566] [info] Loading ./res/colormaps/turbo.json
[2022-05-18 10:07:10.575] [info] Loading ./res/colormaps/viridis.json
[2022-05-18 10:07:10.584] [info] Loading ./res/colormaps/websdr.json
[2022-05-18 10:07:10.606] [info] New DSP samplerate: 4000000.0 (source samplerate is 4000000.0)
[2022-05-18 10:07:10.608] [info] Updating FFT DSP settings: Keep: 65536, Skip: 134464
[2022-05-18 10:07:10.609] [info] HackRFSourceModule 'HackRF Source': Menu Select!
[2022-05-18 10:07:10.609] [info] New DSP samplerate: 4000000.0 (source samplerate is 4000000.0)
[2022-05-18 10:07:10.610] [info] Updating FFT DSP settings: Keep: 65536, Skip: 134464
[2022-05-18 10:07:10.620] [warning] ConfigManager locked, waiting...
[2022-05-18 10:07:10.663] [info] RtAudio stream open
[2022-05-18 10:07:10.682] [info] Updating FFT DSP settings: Keep: 65536, Skip: 134464
[2022-05-18 10:07:10.684] [info] Updating FFT DSP settings: Keep: 65536, Skip: 134464
[2022-05-18 10:07:10.686] [info] Updating FFT DSP settings: Keep: 65536, Skip: 134464
[2022-05-18 10:07:10.688] [info] HackRFSourceModule 'HackRF Source': Tune: 106900000.0!
[2022-05-18 10:07:10.688] [info] Running post-init for Airspy Source
[2022-05-18 10:07:10.688] [info] Running post-init for AirspyHF+ Source
[2022-05-18 10:07:10.689] [info] Running post-init for Audio Sink
[2022-05-18 10:07:10.689] [info] Running post-init for BladeRF Source
[2022-05-18 10:07:10.689] [info] Running post-init for File Source
[2022-05-18 10:07:10.689] [info] Running post-init for Frequency Manager
[2022-05-18 10:07:10.689] [info] Running post-init for HackRF Source
[2022-05-18 10:07:10.689] [info] Running post-init for LimeSDR Source
[2022-05-18 10:07:10.689] [info] Running post-init for Network Sink
[2022-05-18 10:07:10.689] [info] Running post-init for PlutoSDR Source
[2022-05-18 10:07:10.689] [info] Running post-init for RTL-SDR Source
[2022-05-18 10:07:10.690] [info] Running post-init for RTL-TCP Source
[2022-05-18 10:07:10.690] [info] Running post-init for Radio
[2022-05-18 10:07:10.690] [info] Running post-init for Recorder
[2022-05-18 10:07:10.690] [info] Running post-init for Rigctl Server
[2022-05-18 10:07:10.690] [info] Running post-init for SDRplay Source
[2022-05-18 10:07:10.691] [info] Running post-init for SoapySDR Source
[2022-05-18 10:07:10.691] [info] Running post-init for SpyServer Source
[2022-05-18 10:07:10.691] [info] Ready.
[2022-05-18 10:07:21.227] [info] New DSP samplerate: 4000000.0 (source samplerate is 4000000.0)
[2022-05-18 10:07:21.232] [info] Updating FFT DSP settings: Keep: 65536, Skip: 134464
```

diggit, 2022-05-18:
After removal of `soapysdr` module, Sdr++ does not crash anymore.

shuyuan-liu, 2022-05-18:
Glad to see you've got it working.

I'd like to say 1.0.4 is 7 months old and a lot of development has happened since then. Please try the nightly build from https://www.sdrpp.org/nightly/ (this redirects you back to GitHub).

gvanem, 2022-05-18:
> After removal of soapysdr module, Sdr++ does not crash anymore.

I've seen this too where it crashes in libusb. Probably an issue with 2 modules competing for the same USB-device.
(where Hack-RF fails to grab it). Add a `set LIBUSB_DEBUG=3` (or higher) to your environment and watch the libusb events.

diggit, 2022-05-19:
Good to know, there are nightlies available.
79a15ed nightly behaved in same way.

libusb collision could be it, unfortunately, I was not able make libusb print out more info.
my intput to powershell
```powershell
$env:LIBUSB_DEBUG = 3
.\sdrpp.exe -s
```
but there was not extra output in terminal.
As Windows are not my primary platform, I might need some guidance.

gvanem, 2022-05-19:
> `$env:LIBUSB_DEBUG = 3`

Not sure that's legal PWSH-syntax. Try running in `cmd` instead; `set LIBUSB_DEBUG=3 & sdrpp.exe -s`

shuyuan-liu, 2022-05-19:
To clarify the change from `-s` to `-c / --con`, please see this comment and the one below it: https://github.com/AlexandreRouma/SDRPlusPlus/issues/732#issuecomment-1102941099. On 1.0.4, `-s` means "show console (on Windows)"; now it means "run in server mode".

diggit, 2022-05-19:
Here is output, when I set `LIBUSB_DEBUG` to `4`, because `3` did not add much.
[sdrpp_nightly_libusb_dbg4.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/8729090/sdrpp_nightly_libusb_dbg4.txt)

rjp421, 2022-05-25:
possibly related to a HackRF firmware bug https://github.com/greatscottgadgets/hackrf/issues/1068 which i now have after upgrading the firmware to latest release, 2021.03.1.

clicking the Refresh button should re-enable the HackRF if it does not respond.

closed 2022-09-13 by AlexandreRouma