AlexandreRouma/SDRPlusPlus #764: which old macOS version is required ?

Observerist, 2022-05-22:
sdr++ needs at least which macOS version ? 

Mojave is not compatible. That I could see.
Which following ones are OK then... Catalina, Big Sur ?
Or only Monterey ? ?

Thx

ja-Bertolin, 2022-05-22:
I am using BigSur and it works great, with the last nightly build installed the crash problem was over.

Observerist, 2022-05-22:
thank you ja-Bertolin, 

Can anyone confirm sdr++ being "allowed" on Catalina ? 
Under which macOS was the app's binary built ?
That usually validates the working OS and eliminates older versions.

In the PC world of Ham & SWL apps, older computer are usually not thrown away, but used for specific purposes.
That gives a new life to PCs. 
As an avid macOS user, I regret that software is developed only around the latest OS versions.

billybobnn5, 2022-05-22:
All, I am running v1.04 on Catalina on a late 2012 I7 mac mini. I am using it as panadptor for my TS590S/Airspy HF+ rig with the n2iw flrig cat-relay fork mentioned here about 10days ago. My old choice with gqrx-hamlib scripts was always crashing and loosing connections. But this solution is much more stable. sdr++ can work indeed on mac and is a great panadaptor with available GIT software code.

Observerist, 2022-05-22:
Thx billybob for the Catalina confirmation.

Then I'll be able to retrieve a "retired" 2014 MacMini and keep Catalina on it.

The follow-panadapter function also crossed my mind... remote controlling an R75, like in the old days with HRD on an old PC.

Thanks again.

AlexandreRouma, 2022-06-22:
I know people who have compiled it fine on high sierra. The reason  the automated builds only run on Catalina or more is that's what it was compiled on, and apple doesn't allow running on older versions than the compiler.
Your only option is to compile from source

closed 2022-06-22 by AlexandreRouma